microsoft/AirSim #1: Build instructions lead to #include error on Eigen/Dense

jwusch, 2017-02-15:
Following build instructions from [here](https://github.com/Microsoft/AirSim/blob/master/docs/build.md)  gives me an error inside of VS2015  "cannot open source file "Eigen/Dense"	AirLib	c:\boost_1_63_0\AirSim\AirSim\AirLib\include\common\VectorMath.hpp	12	

I have set EIGEN_HOME env variable, and manually added that to the project's "Include Directories" under the property panel, but no joy.

clovett, 2017-02-15:
Ah, sorry about that, perhaps we need to fix the docs, the variable should be named EIGEN_ROOT.

jwusch, 2017-02-15:
Ah sorry my bad.  I did call it EIGEN_ROOT, but have HOME on the brain for some reason.  Still not working though.

jwusch, 2017-02-15:
So searching through the source, I don't seen any references to EIGEN_ROOT, within the source code.  I am assuming that path should be referenced in the "Include Directores" right?

clovett, 2017-02-15:
It will be in the make files, either cmake or .vcxproj.  It looks for eigen3 folder inside EIGEN_ROOT, so if you removed that you may need to put that back, and you do not include eigen3 in the   EIGEN_ROOT.

NicoJuicy, 2017-02-15:
I had the same problem.

 - Cloned the project locally
 - Downloaded Eigen and Boost
 - Installed boost ( running the bat file)
 - Run the *.sln on Windows ( VS 2015), building it.

Try to run the test "HelloDrone"

Error dissappeared, as the solution finds the required libraries :)

sytelus, 2017-02-16:
As long as you have environment variable `EIGEN_ROOT` that points to the *root* folder of Eigen library installation, you shouldn't have a problem. The root of Eigen library would should have child folder called `eigen3`.

lovettchris, 2017-02-17:
I also tried to make the Eigen install instructions more clear, see [install_eigen.md](https://github.com/Microsoft/AirSim/blob/master/docs/install_eigen.md).

morrisonbrett, 2017-02-20:
I ran into this issue today while setting up.  I modified the docs for both Eigen and Boost to be clearer and created a PR: https://github.com/Microsoft/AirSim/pull/23

closed 2017-02-25 by sytelus

AloshkaD, 2018-06-24:
I had this issue on win 10 and  UE4.18 . Inside AirSim\AirLib\deps there are two folders, eign-eign- and eign3. Tha latter was empty so I coppied the contents of eign-eign- into eign3 and the compilation worked

yejuns, 2020-05-12:
when i meet this question, i try to open the script file 'build.cmd', according to the contents of this script file 
IF NOT EXIST AirLib\deps\eigen3 (
    powershell -command "& { [Net.ServicePointManager]::SecurityProtocol = [Net.SecurityProtocolType]::Tls12; iwr https://gitlab.com/libeigen/eigen/-/archive/3.3.7/eigen-3.3.7.zip -OutFile eigen3.zip }"
    powershell -command "& { Expand-Archive -Path eigen3.zip -DestinationPath AirLib\deps }"
    powershell -command "& { Move-Item -Path AirLib\deps\eigen* -Destination AirLib\deps\del_eigen }"
We can know we should to download Eigen, then decompression the package in AirLib\deps\eigen,
this is fix my problem.

denmonz, 2021-02-12:
> I ran into this issue today while setting up. I modified the docs for both Eigen and Boost to be clearer and created a PR: #23

The link you reference to download Eigen does not work anymore. Site says, "Repository eigen/eigen not found."

M1NDX, 2021-02-13:
I had the same issue:   first "Repository eigen/eigen not found."  but the link is correct,  I think you need to open the Developer prompt as admin   

In order to solve the problem of   "Eigen/Dense" error  I  copied all files of eigen-3.3.7  to eigen3    as indicated by yejuns here [https://github.com/microsoft/AirSim/issues/1#issuecomment-627275245](url)

lovettchris, 2021-02-15:
This is done for you in [build.cmd](https://github.com/microsoft/AirSim/blob/master/build.cmd) on Windows or [build.sh](http://github.com/microsoft/AirSim/blob/master/build.sh) on Linux.

HassanNawazish1, 2021-04-29:
Replace #include <Eigen/Dense> or #include <Dense>
with
#include <eigen3/Eigen/Dense> if you are using ubuntu

Gastastrophe, 2021-10-02:
If anyone is still having this issue, I found that I only had an empty eigen3 folder and no eigen-eigen- folder. I had to grab the contents of an old installation that worked and put it inside eigen3 to fix it. I've attached the folder I placed inside eigen3 here.
[Eigen.zip](https://github.com/microsoft/AirSim/files/7272228/Eigen.zip)
.