microsoft/AirSim #2: Does it have mac support

idoan, 2017-02-16:
I couldn't see Mac listen on read.me, so I wanted to be sure. does it have mac support ?

lovettchris, 2017-02-16:
Not yet.  Unfortunately Apple doesn't let us run a free version of their OS in a virtual machine, and as you can imagine there are not too many spare Mac computers around Microsoft for us to test on.  But given that the Mac kernel is a close cousin to Linux, it shouldn't be that hard to get the Linux build working on Mac, right?  There are some cmake files in the cmake folder that should get you going.  Not sure what the translation is from gcc command line args to llvm though.  We would just add another if (MAC) block in the CommonSetup.cmake file next to if (UNIX) and if (Win32).  MavLinkCom depends on boost, I assume that works on Mac, and Eigen.  Then there's one more spot where you'd need to implement "createdirectory", although the Unix code "mkdir" should work there, and there's a place where we'd like to find the user's Document folder to store our settings.json file, not sure what the convention on Mac is for that, but on unix we just use ~/Documents/AirSim.  other than that the rest of the code is full cross platform except some of our tools (like LogViewer).  But there are other tools that can view mavlink logs on Mac already.

MAVProxyUser, 2017-02-16:
"you can imagine there are not too many spare Mac computers around Microsoft for us to test on" stop on by the Office X team ;) Should be plenty there.

and7ey, 2017-02-16:
Would appreciate if somebody shares compiled version

d13-kevinf, 2017-02-16:
Here is the first hurdle...

$ brew install eigen@3.2
$ cmake . -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/eigen@3.2/3.2.10/
-- Running CommonSetup...
-- found AIRSIM_ROOT=/Users/kfinisterre/Desktop/AirSim
-- Found Eigen3: /usr/local/Cellar/eigen@3.2/3.2.10/include/eigen3 (Required is at least version "3") 
-- Running CommonSetup...
-- found AIRSIM_ROOT=/Users/kfinisterre/Desktop/AirSim
-- Boost version: 1.63.0
-- Found the following Boost libraries:
--   system
--   filesystem
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Running CommonSetup...
-- found AIRSIM_ROOT=/Users/kfinisterre/Desktop/AirSim
-- Boost_INCLUDE_DIR=/usr/local/include
-- Boost version: 1.63.0
-- Found the following Boost libraries:
--   system
-- Running CommonSetup...
-- found AIRSIM_ROOT=/Users/kfinisterre/Desktop/AirSim
-- Boost version: 1.63.0
-- Found the following Boost libraries:
--   system
--   filesystem
-- Running CommonSetup...
-- found AIRSIM_ROOT=/Users/kfinisterre/Desktop/AirSim
-- Boost version: 1.63.0
-- Found the following Boost libraries:
--   system
--   filesystem
-- Running CommonSetup...
-- found AIRSIM_ROOT=/Users/kfinisterre/Desktop/AirSim
-- Boost version: 1.63.0
-- Found the following Boost libraries:
--   system
--   filesystem
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/kfinisterre/Desktop/AirSim/cmake
Kevins-MBP:cmake kfinisterre$ make 
Scanning dependencies of target AirLib
[  2%] Building CXX object AirLib/CMakeFiles/AirLib.dir/Users/kfinisterre/Desktop/AirSim/AirLib/src/control/DroneControlBase.cpp.o
clang: error: -Wl,--no-as-needed: 'linker' input unused [-Werror,-Wunused-command-line-argument]
clang: error: -ldl: 'linker' input unused [-Werror,-Wunused-command-line-argument]
clang: error: -ldl: 'linker' input unused [-Werror,-Wunused-command-line-argument]
make[2]: *** [AirLib/CMakeFiles/AirLib.dir/Users/kfinisterre/Desktop/AirSim/AirLib/src/control/DroneControlBase.cpp.o] Error 1
make[1]: *** [AirLib/CMakeFiles/AirLib.dir/all] Error 2
make: *** [all] Error 2

lovettchris, 2017-02-16:
That's cool, do you need to edit the CommonSetup.cmake file to modify the CMAKE_CXX_FLAGS so llvm is happy ?

d13-kevinf, 2017-02-16:
Yeah this gets me a little further:

diff --git a/cmake/cmake-modules/CommonSetup.cmake b/cmake/cmake-modules/CommonSetup.cmake
index 75df3a3..3757f7d 100644
--- a/cmake/cmake-modules/CommonSetup.cmake
+++ b/cmake/cmake-modules/CommonSetup.cmake
@@ -14,7 +14,7 @@ macro(CommonSetup)
         set(BOOST_OVERRIDES " -Wno-error=undef  -Wno-error=ctor-dtor-privacy -Wno-error=old-style-cast  -Wno-error=shadow -Wno-error=redundant-decls -Wno-error=missing-field-initializers  -Wno-error=unused-parameter") 
         ## Mavlink requires turning off -pedantic  and -Wno-error=switch-default 
         set(MAVLINK_OVERRIDES "-Wno-error=switch-default ") 
-        set(CMAKE_CXX_FLAGS "-std=c++14 -ggdb -Wall -Wextra  -Wstrict-aliasing -Werror -fmax-errors=2 -Wunreachable-code -Wcast-qual -Wctor-dtor-privacy -Wdisabled-optimization -Wformat=2 -Winit-self -Wmissing-include-dirs -Wnoexcept -Wold-style-cast -Woverloaded-virtual -Wredundant-decls -Wshadow -Wstrict-null-sentinel -Wstrict-overflow=5 -Wswitch-default -Wundef -Wno-unused -Wno-variadic-macros -Wno-parentheses -fdiagnostics-show-option ${MAVLINK_OVERRIDES} ${BOOST_OVERRIDES}  -Wl,--no-as-needed -ldl -ldl ${CMAKE_CXX_FLAGS}")
+        set(CMAKE_CXX_FLAGS "-std=c++14 -ggdb -Wall -Wextra  -Wstrict-aliasing -Werror  -Wunreachable-code -Wcast-qual -Wctor-dtor-privacy -Wdisabled-optimization -Wformat=2 -Winit-self -Wmissing-include-dirs -Wold-style-cast -Woverloaded-virtual -Wredundant-decls -Wshadow -Wstrict-overflow=5 -Wswitch-default -Wundef -Wno-unused -Wno-variadic-macros -Wno-parentheses -fdiagnostics-show-option ${MAVLINK_OVERRIDES} ${BOOST_OVERRIDES} ${CMAKE_CXX_FLAGS}")
         set(BUILD_PLATFORM "x64")
 
     ELSE()

sytelus, 2017-02-16:
Right now we are seeing some odd behaviour in GCC but theoretically our C++ code is designed to be cross platform so it should be compilable on Mac. Unreal also works on Mac so theoretically you should be able to make things work. A limiting factor, however, would be GPU capacity on Mac. Even moderate scale environments require 4GB GPU RAM. I highly doubt underpowered MBPs would cut it.

idoan, 2017-02-16:
so, we need to have strong GPU (like nvidia 1070 or so?) to use this software ? integrated graphics has no luck then ? gosh :(

can you specify the minimum requirements on the readMe ? thanks

MAVProxyUser, 2017-02-16:
https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md states:"STATUS: all of the MavLinkCom library builds and runs on Linux, but the AirLib code is running into a weird difference in how GCC implements some -std=c++14 features... stay tuned..." 

I suspect this impacts Mac users as well FWIW.

lovettchris, 2017-02-18:
Thanks to Will Sackfield, the OSX build now works.  See [commit details](https://github.com/Microsoft/AirSim/pull/15).

d13-kevinf, 2017-02-18:
I still get an error... 

1 warning generated.
[ 97%] Linking CXX executable MavLinkTest
[ 97%] Built target MavLinkTest
1 warning generated.
[100%] Linking CXX executable DroneShell
ld: warning: ignoring file /Users/kfinisterre/Desktop/AirSim/AirLib/deps/rpclib/lib/x64/linux/librpc.a, file was built for archive which is not the architecture being linked (x86_64): /Users/kfinisterre/Desktop/AirSim/AirLib/deps/rpclib/lib/x64/linux/librpc.a
Undefined symbols for architecture x86_64:
  "rpc::client::get_next_call_idx()", referenced from:
      std::__1::future<clmdep_msgpack::v1::object_handle> rpc::client::async_call<bool>(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) in libAirLib.a(RpcLibClient.cpp.o)
      std::__1::future<clmdep_msgpack::v1::object_handle> rpc::client::async_call<>(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in libAirLib.a(RpcLibClient.cpp.o)

clovett, 2017-02-18:
Ah, so we need a mac build for the GitHub "rpclib" library then...

lovettchris, 2017-02-18:
Now that rpclib is a git submodule, is this fixed?

lovettchris, 2017-02-18:
there is a pull request that moves rpclib to git submodule, which is the right fix for OSX.

d13-kevinf, 2017-02-19:
@clovett  This should help you get going... I have the manual steps to try to PR (before it gets merged) https://github.com/Microsoft/AirSim/pull/16#issuecomment-280828193

sytelus, 2017-02-25:
I've merged PR16. Let's continue this discussion at https://github.com/Microsoft/AirSim/issues/36. I'll close this issue as duplicate for now.

closed 2017-02-25 by sytelus

ghexp, 2017-03-08:
@lovettchris, WRT [Apple doesn't let us run a free version of their OS in a virtual machine](https://github.com/Microsoft/AirSim/issues/2#issuecomment-280251875)), it looks like Travis CI offers OS X build environments with XCode (up to and including 10.12 and XCode 8.2):
https://docs.travis-ci.com/user/osx-ci-environment/#Compilers-and-Build-toolchain
Since this is a public github project, that service is free. I've started an [example yml file](https://github.com/ghexp/AirSim/blob/master/.travis.yml) that is a work in progress ([build not successful](https://travis-ci.org/ghexp/AirSim)). I can make a pull request when it's building or make one now and let more people help fix it.  Getting/keeping this working would go along way to help address #2, #15, #36, #18,  #19, #43, #72

d13-kevinf, 2017-03-08:
In the OpenPilot project we used to have trusted project members run the OSX build agents for Atlassian toolset on our machines. Can we do that for you in this case? Offer up a legit machine agent for your builds ?

inkjet, 2017-03-11:
@ghexp That is very cool - didn't know that option was available in Travis CI.  Would love to see AirSim working right out of the box on a Mac - I know several people who would be interested immediately.

raymondchua, 2017-10-03:
Can anyone share the steps to get AirSim running in MacOS?

DJISDKUser, 2017-10-03:
I gave up a while ago... has anything changed recently?

thiago-cavalcante, 2021-06-16:
Does AirSim run on M1 mac? Or in macOS version greater than Catalina (10.5)?