microsoft/AirSim #3: Datalog

marcelinomalmeidan, 2017-02-16:
Hi, 

I am interested in using this simulation framework with my research. However, I would like to know more about what kind of data that we can acquire: is is only simulated sensor data or do we also have access to "truth" data?

Thanks!

lovettchris, 2017-02-16:
without writing any code you can get the mavlink streams.  The [Log Viewer ](https://github.com/Microsoft/AirSim/blob/master/docs/log_viewer.md) is one way to visualize those streams.  If you want to write some C++ code you can also call the DroneControlBase API to get all the vehicle state, and Scene, Depth or Segmentation image streams.

marcelinomalmeidan, 2017-02-16:
@lovettchris, thank you for your answer!
I checked the Log Viewer page, which should provide "noisy" sensor simulated data, right? 
Otherwise, when you mention the "vehicle state" from the API, would this be real, non-noise-corrupted data? I am interested in having "truth" for attitude and position, not only the noisy/filtered data that Mavlink typically provides.

Thanks!

sytelus, 2017-02-16:
All of below data is possible to retrieve:

- Ground truth linear and angular position, velocity, acceleration
- Ground truth mono or stereo image, depth image and object segmentation image
- Sensor data for barometer, magnetometer, gyroscope, accelerometer and GPS
- Rotor control signals received from flight controller

Right now our API already supports image and pose data. It's fairly trivial to add other data as well.

closed 2017-02-16 by marcelinomalmeidan

ganeshrk95, 2018-07-04:
Is it possible to get the RC raw inputs when in HIL mode?