microsoft/AirSim #6: Compile Error chrono(769): error C2679

thias15, 2017-02-16:
Getting the following error when building. 
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\chrono(769): error C2679: binary '+=': no operator found which takes a right-hand operand of type 'const std::chrono::duration<double,std::ratio<1,1>>' (or there is no acceptable conversion)

sytelus, 2017-02-16:
Could you please confirm you are using Visual Studio 2015 Update 3 and comping from VS2015 x64 Native Tools Command Prompt?

thias15, 2017-02-16:
I had VS2015, Update1. After updating to Update 3 it compiled. Thanks!
I do find it a bit strange though that it only compiles with this specific version.

lovettchris, 2017-02-18:
sounds like this is resolved now.

closed 2017-02-18 by lovettchris