microsoft/AirSim #9: .uproject file Generate Visual Studio Project files compile error

RizHuang, 2017-02-17:
After I had followed the instructions , everything went successfully . But when I re-generated the .sln e Visual Studio Project files then compile in the VS . It show me some problems in the "PIPCamera.h" file .
`Invalid BlueprintType enum base - currently only uint8 supported	demo`
`UnrealHeaderTool failed for target 'demoEditor' (platform: Win64, module info: C:\Users\RizHuang\Documents\Unreal Projects\demo\Intermediate\Build\Win64\demoEditor\Development\demoEditor.uhtmanifest, exit code: OtherCompilationError (5)).`	

My VS is 2015 and update 3. and all the software mentions in the instruction doc are successfully installed .

Any suggestions are thankful !

RizHuang, 2017-02-17:
And also the error code # MSB3075

lovettchris, 2017-02-17:
Did this fix it already? [Merge pull request #7](https://github.com/Microsoft/AirSim/commit/2f1b14ba7e5a18388b888046c38379e2935a28c4) from gawela/fix/enum.

sytelus, 2017-02-18:
Which Unreal version are you using? The latest came out yesterday (4.15) and that one we haven't tested yet but version 4.14 should be fine (which is what we are using for testing).

RizHuang, 2017-02-19:
Yes  I comiled successfully on 4.14

RizHuang, 2017-02-19:
Thank you very much .

closed 2017-02-19 by RizHuang