microsoft/AirSim #14: Build error - Utils.hpp and UpdatableObject.hpp not found

jediMasterMindTrick, 2017-02-18:
I've tried to add the plugin to different projects with different maps, but no luck.  Anyone have any ideas?

[image]

lovettchris, 2017-02-18:
I just tried a fresh test with new Unreal project, and it works, can you check the setup setups in [this video](https://www.youtube.com/watch?v=HNWdYrtw3f0&feature=youtu.be)? Thanks.

sytelus, 2017-02-18:
Plugin is dependent on setting up include and library paths. You might want to make sure your new project has same settings for these as our projects.

jediMasterMindTrick, 2017-02-20:
> I just tried a fresh test with new Unreal project, and it works, can you check the setup setups in this video? Thanks.

Do I need to have my hardware connected when I'm compiling the Unreal project with the AirSim library?

> Plugin is dependent on setting up include and library paths. You might want to make sure your new project has same settings for these as our projects.

I followed the instructions in the image below and was able to get the AirSim library to compile, but once added to an Unreal project, I cannot get the project itself to compile.  Are there other things I need to include or add path variables for?
[image]

lovettchris, 2017-02-22:
Did you right click and re-"generate visual studio project files" after adding the Plugin folder?  We have some build instructions that need to be picked up by the unreal generation process.

netpearce, 2017-02-24:
I had the same issue -- essentially AirLib includes etc weren't copied into the Unreal\Plugins folder (along with some library objects for linking) when I compiled the main AirLib code.

When I first installed I somehow skipped the step to run build.cmd and compiled directly via the solution file 

Going back and following this step solved the issue.

sytelus, 2017-02-25:
Please reopen if issue still exists. I've verified steps as [documented here](https://github.com/Microsoft/AirSim/blob/master/docs/build.md) and they work on fresh install as long as [prerequisites](https://github.com/Microsoft/AirSim/blob/master/docs/prereq.md) are met.

closed 2017-02-25 by sytelus