microsoft/AirSim #17: what are the system requirements ?

idoan, 2017-02-18:
I read an answer from your teammember on another issue that this simulator requires discrete gpu with 4gb or more ram. does this mean I can't run this on pc with integrated graphics (hd 530) on low settings ? likewise, what are the other hardware requirements that might be critical for simulator's performance ?

thanks.

sytelus, 2017-02-18:
It depends on what Unreal environment you use. For Neighbourhood Pack described in docs, a discrete GPU with minimum of 4GB RAM is a must. As environment becomes bigger you need more. For example, running "Boy with Kite" environment would not go very well unless you have card with 8GB RAM. There are some low detailed environments available in Unreal market that might work on integrated graphics but we haven't tested it. 

Apart from this, we also use high frequency physics engine which requires its own CPU core to run it. So having at least quad core machine helps.

Our current focus is AI research and typically folks in this field work on beefy desktops so we haven't put a lot of effort on making things work on lower specs. I was able to run AirSim on my Lenovo P50 desktop that has discrete card at low res and I do development on that machine sometime. We do wish to make "Consumer Edition"  of AirSim down the line that doesn't require Pixhawk or RC and can run on less powerful machines such as Macbook Pro. If you are developer and would like to contribute, please do feel free to help us out :).

closed 2017-02-19 by sytelus

andre-nguyen, 2018-08-02:
@sytelus Hi there, I know this is an old issue but I would like to know what the specs are on your dev machine so I can know what I need to run this as smooth as possible.

TritonSailor, 2018-08-28:
@sytelus Were you able to package BoyandHisKite maps with airsim gamemode?  I've got it working in the editor, runtime errors with the packaged .exe