microsoft/AirSim #18: How to download Unreal Environment in Linux

yanpanlau, 2017-02-18:
Hi, I installed unreal engine 4.0 in ubuntu 14.04 but I cannot download the Unreal Environment in Linux. (Modular Neighborhood Pack or Infinity Blade series). 

I also checked out the following link to download some sample environment but all the links are broken

https://wiki.unrealengine.com/Linux_Demos

Can you advice how to download various environment in linux? --Thanks

sytelus, 2017-02-19:
To download marketplace content you need Unreal App Launcher. Unfortunately [Epic has not released ](https://wiki.unrealengine.com/Linux_Support) it for Linux and there are no announcement for releasing it in near term. 

Usually, the most convenient way is to create your Unreal project in Windows, generate all files and then copy them over to Linux machine. If you only have Linux machine, you can use probably use VM to do this but there are no other workarounds for this AFAIK.

Here are the steps to run Unreal on Linux: https://wiki.unrealengine.com/Running_On_Linux

sytelus, 2017-07-27:
We been busy fixing various issues in Linux build. Now we have new code out that should build and run on Ubuntu 16 with few known issues. Please see [docs](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md)) for instructions.

closed 2017-07-27 by sytelus