microsoft/AirSim #19: Build error in linux

NirojPokhrel, 2017-02-18:
When trying to build Airsim in linux, I can build till 88% and get an error.
[ 88%] Building CXX object HelloDrone/CMakeFiles/HelloDrone.dir/home/niroj/Airsim/AirSim/HelloDrone/main.cpp.o

/home/niroj/Airsim/AirSim/AirLib/deps/rpclib/include/rpc/msgpack/object_fwd.hpp:94:8: warning: all member functions in class ‘clmdep_msgpack::v1::has_as<T>’ are private [-Wctor-dtor-privacy]
 struct has_as {
        ^~~~~~
/home/niroj/Airsim/AirSim/HelloDrone/main.cpp: In function ‘int main()’:
/home/niroj/Airsim/AirSim/HelloDrone/main.cpp:94:27: error: use of deleted function ‘rpc::rpc_error::rpc_error(const rpc::rpc_error&)’
     catch (rpc::rpc_error  e) {
                           ^
In file included from /home/niroj/Airsim/AirSim/HelloDrone/main.cpp:12:0:
/home/niroj/Airsim/AirSim/AirLib/deps/rpclib/include/rpc/rpc_error.h:18:7: note: ‘rpc::rpc_error::rpc_error(const rpc::rpc_error&)’ is implicitly deleted because the default definition would be ill-formed:
 class rpc_error : public std::runtime_error {
       ^~~~~~~~~
/home/niroj/Airsim/AirSim/AirLib/deps/rpclib/include/rpc/rpc_error.h:18:7: error: use of deleted function ‘clmdep_msgpack::v1::object_handle::object_handle(const clmdep_msgpack::v1::object_handle&)’
compilation terminated due to -fmax-errors=2.

lovettchris, 2017-02-18:
Can you follow the [linux build setup steps](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md) to upgrade to GCC 6 ?

NirojPokhrel, 2017-02-19:
Hi chris,
I followed the instructions and installed the gcc 6 but still getting the same error? Have you installed it in Linux? Was changing to GCC 6 was enough for you?

lovettchris, 2017-02-22:
Yes, I have verified clean linux build using GCC 6 on multiple machines.  Can you try the build.sh script I added to the cmake folder ?

NirojPokhrel, 2017-02-22:
Thank you for your help.
I was still getting error regarding rpc. So, I downloaded the rpclib from https://github.com/rpclib/rpclib
and replaced the library and I was able to compile it successfully. 
But it would be nice if the future release of Airsim has librpc.a library that can be compiled from the source code in ther project itself.

Again, Thank you for your support I will close this issue.

closed 2017-02-22 by NirojPokhrel