microsoft/AirSim #20: MSBuild

sslaughter, 2017-02-18:
I apologize if this is not the appropriate place for support questions, but I've run into an issue while trying to install. When trying to build airsim (build.cmd in repo) I run into the error of it being unable to find microsoft.build.tasks.v12.0.dll. It's looking in the following directory: 

program files (x86)\MSBuild\Microsoft\14.0\bin

The file is not there and after looking around I went ahead and installed MSBuild separately from [here](https://msdn.microsoft.com/en-us/library/hh162058.aspx). This populated the ....\MSBuild\Microsoft\12.0 directory and the missing dll is there. However, I'm uncertain how to redirect the MSBuildToolsPath that is referenced. This may be simple, but my search has not yet found a good way of doing so that I understand.

Full error message:

C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v14.0\CodeAnalysis\Microsoft.CodeAnalysis
.targets(219,5): error MSB4175: The task factory "CodeTaskFactory" could not be loaded from the a
ssembly "C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Build.Tasks.v12.0.dll". Could not load
 file or assembly 'file:///C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Build.Tasks.v12.0.dl
l' or one of its dependencies. The system cannot find the file specified. [D:\Epic Games\AirSim\M
avLinkCom\MavLinkCom.vcxproj]

System Info:
Windows 10, 64bit
Visual Studio Community 15

lovettchris, 2017-02-18:
Ah yes, there is a dependency in our .vcxproj files on VS 2015 version of C++, you will see that in this :
  <PlatformToolset>v140</PlatformToolset>.  Notice our [build.md ](https://github.com/Microsoft/AirSim/blob/master/docs/build.md)instructions mention VS2015.  We have not tested on VS 2013.  But if you want to try it you could try the [cmake instructions](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md), scroll to the bottom to see how to build new .vcxproj files on windows.

sytelus, 2017-02-25:
AirSim most likely won't work below VS2015 Update 3 because it needs few C++14 features to compile rpclib.

closed 2017-02-25 by sytelus