microsoft/AirSim #26: no ‘object' file generated

LeadgerLee, 2017-02-20:
Hello, everyone. When I build the AirSim project, it displayed some errors while compiling. As follow, 
----------------------------------------------------------------------------------------------------------------
  d:\code\eigen3\eigen\src/Core/arch/CUDA/Half.h : error C2220: warning treated as error - no 'object' file generated
[D:\GitHub\repo\AirSim\AirLib\AirLib.vcxproj]
  d:\code\eigen3\eigen\src/Core/arch/CUDA/Half.h : error C2220: warning treated as error - no 'object' file generated
... etc...
-----------------------------------------------------------------------------------------------------------------

Should I change the option "Treat warning as error" from "Yes(/WX)" to "No(/WX-)" in the property page of project?
Thank you very much.

sytelus, 2017-02-20:
Which OS are you using? Are you compiling using build command? Also please post more error details.

jy723, 2017-02-21:
I've got the same problem.My raphics card is NVDIA GeForce GTX745.

fmzhu, 2017-02-21:
I've got the same problem. My OS ist Windows10 and the Visual Studio is 2015. I compile use build command.

jy723, 2017-02-21:
error message: half.h : warning C4819: error936
The file eigen3\Eigen\src\Core\arch\CUDA\half.h is saves as Unicode－1200,the warnings disapperance;

fmzhu, 2017-02-21:
@jy723 thank you very much. The problem is fixed.

LeadgerLee, 2017-02-21:
@jy723 Thank you very much. The problem is fixed.

closed 2017-02-21 by LeadgerLee

spencerh-b, 2017-10-11:
@LeadgerLee can you explain what exactly you did to fix this?

Jackyjae, 2017-12-07:
@jy723 I met the same problem today. How can I fix it?

bys1123, 2018-04-07:
How is this happen? Everyone have this issue? Is this file can be fixed? @jy723

kingium, 2018-07-15:
I get nine C2220 errors with `/deps/eigen3/eigen/src/core/arch/cuda/half.h`, each at:
-  `RpcLibClientBase.cpp`
-  `RpcLibServerBase.cpp`
-  `MultirotorApiBase.cpp`
-  `ObstacleMap.cpp`
-  `SafetyEval.cpp`
-  `CarRpcLibClient.cpp`
-  `CarRpcLibServer.cpp`
-  `MultirotorRpcLibClient.cpp`
-  `MultirotorRpcLibServer.cpp`