microsoft/AirSim #27: Can't take off

ivanqiu2, 2017-02-20:
I'm able to arm and watch the props spin up but throttling up on my transmitter does nothing. I have tried calibrating the RC inputs through QGroundControl multiple times as well as try 2 different transmitter/receiver combos. 

Also disarming doesn't seem to stop the props from spinning.

sytelus, 2017-02-20:
Try press R. After few seconds of delay it would show debug window. There you can see rotor controls command transmitter is sending. You might want to take a screenshot and post a link here.

Yes, disarming doesn't stop rotor right now. This is both a bug in PX4 as well as our simulator. PX4 should send rotor 0 control signals after disarming but it doesn't. And we should probably zero out our rotor controls once disarming is detected. We are planning to do fix for later.

ivanqiu2, 2017-02-20:
[image]

sytelus, 2017-02-22:
This is strange. Could you please check your RC control calibration in QGroundControl. Only thing I can think of is that your RC is sending value too low even at max setting. You might want to check any related parameters in QGroundControl.

LeadgerLee, 2017-02-22:
@ivanqiu2 Hello! Have you solved the problem? I have encountered the same situation as you.

ivanqiu2, 2017-02-22:
I forgot to mention an important detail. I was initially using the Pixracer V1 from AUAV in which I encountered this problem. However when I tried with the Pixhawk everything seemed to work correctly. I still have not been able to get it running with the Pixracer which is what I will be needing HIL for.

lovettchris, 2017-02-23:
Are you using version 1.4.4 of the PX4 firmware on the Pixracer?

ivanqiu2, 2017-02-23:
Yes, I'm using the 1.4.4 V4 firmware that was linked in prerequisite.

lovettchris, 2017-02-23:
Can you get the latest and try again.  This [important fix ](https://github.com/Microsoft/AirSim/commit/5685faa22f5e0dda3237e4bd9b1c0403225e362b) might help...

LeadgerLee, 2017-02-23:
@lovettchris At first I use Pixracer and it armed and couldn't take off. Just now I try a pixhawk and it likely work well.

ivanqiu2, 2017-02-23:
I pulled the new code, re-compiled, and tried both 1.4.4 and the latest firmware with no success using the Pixracer. Pixhawk still works.

sytelus, 2017-02-25:
Thanks for trying Pixracer. I've changed the title and we'll keep this issue open. If you find reason/solution, please feel free to let us know/contribute :).

jamuus, 2017-03-03:
Any clues debugging this? I'm also trying to get Pixracer to work.

clovett, 2017-03-03:
You probably need to specify the com port name instead of using auto discovery (*) in the -/Documents/AirSim/setting.json file.

jamuus, 2017-03-03:
I can connect and am receiving logging output from the device into the game. I can communicate with the controller using DroneShell and can arm/disarm. The problem at the moment seems to be the board rejecting transitioning to OFFBOARD mode. I think I briefly had it in the right mode but even then the in game quad remained unresponsive and the takeoff command would timeout.

jjy201314, 2017-03-09:
RPC Error: rpclib: function 'takeoff' (called with 1 arg(s)) threw an exception. The exception contained this information: TakeOff command rejected by drone.
can't takeoff ,my error above .

lovettchris, 2017-03-16:
Ok, we've moved to the latest PX4 firmware, and we have our own fork of that now so we can fix bugs in there more easily, see [https://github.com/lovettchris/firmware](https://github.com/lovettchris/firmware).  I'm also planning to add more verbose output that explains why certain state transitions are being denied which should help debug this.

bellahduan, 2017-05-19:
how to set the settings.json for AirSim

bellahduan, 2017-05-19:
how to set the settings.json from AirSim

lovettchris, 2017-06-10:
I now have a Pixracer controller, and I'm happy to report it works great with AirSim.

closed 2017-06-10 by lovettchris