microsoft/AirSim #30: Access violation when specifiying json settings

mape1082, 2017-02-20:
Hi, 
Since I don't have the physical RC, I am trying to use the firware in Ubuntu guest, on host windows 10.

In Ubuntu I get:
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..

Therefore I try running the simulation in W10. Assign the game mode to SimGameMode, then Play button and then I get a  crash. The log contains errors such as the following:

> [2017.02.20-18.41.16:412][746]PIE:Error: Error Blueprint Runtime Error: Accessed None trying to read property ExternalCamera from function: 'ExecuteUbergraph_BP_CameraDirector' from node: Set CameraStartLocation in graph: Set Initial Offset in object: BP_CameraDirector with description: Accessed None trying to read property ExternalCamera
[2017.02.20-18.41.16:412][746]LogScript:Warning: Accessed None trying to read property ExternalCamera
	BP_CameraDirector_C /Temp/UEDPIE_0_Untitled_1.Untitled_1:PersistentLevel.BP_CameraDirector_C_0
	Function /AirSim/Blueprints/BP_CameraDirector.BP_CameraDirector_C:ExecuteUbergraph_BP_CameraDirector:0075

My settings.json file:

`{
  "AirControlCompID": 1,
  "AirControlSysID": 134,
  "ExtRendererCompID": 1,
  "ExtRendererSysID": 167,
  "ExternalSimPort": 14588,
  "LocalHostIp": "127.0.0.1",
  "LogViewerHostIp": "127.0.0.1",
  "LogViewerPort": 14388,
  "Pixhawk": {
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "UdpIp": "my_ubuntu_vm_ip_address",
    "UdpPort": 14560,
    "UseSerial": true
  },
  "QgcHostIp": "127.0.0.1",
  "QgcPort": 14550,
  "SimCompID": 42,
  "SimSysID": 142
}`

replace my_ubuntu_vm_ip_address with the IP address of my VM.

I tried commenting out all of the pairs except the Pixhaw and still crashes.

Edit:
I am now getting aqccess violation in  >	UE4Editor-AirSim.dll!mavlinkcom::MavLinkConnection::join(std::shared_ptr<mavlinkcom::MavLinkConnection> remote, bool) Line 109	C++

When I dont specify json file, it goes ok, but it wont connect to PX4 on Ubuntu. If I specify json settings, wit seems so far no matter what, it will crash.

Any ideas?

Thanks and regards.

lovettchris, 2017-02-20:
I just checked in [this fix](https://github.com/Microsoft/AirSim/commit/3a2f045d57bd228b9aa59eeba1dc2f9ccf0e8b5d).

If you want to run the above posix_sitl in a VirtualBox Ubuntu machine then it will have a different ip address from localhost. So in this case you need to do the following: 1. Run the sitl so it is "listening", you should see the message INFO [simulator] Waiting for initial data on UDP port 14560 2. Edit the settings file so that the UseSerial is false, the UdpIp address is the ip address of your virtual machine and the LocalIpAddress is the address of your host machine running the Unreal engine.

mape1082, 2017-02-21:
Thanks a lot.

closed 2017-02-21 by mape1082