microsoft/AirSim #32: cannot open file 'libboost_filesystem-vc140-mt-gd-1_63.lib'

Zhao1ming, 2017-02-20:
Here is the main error: 

LINK : fatal error LNK1104: cannot open file 'libboost_filesystem-vc140-mt-gd-1_63.lib' [D:\AirSim\Air\AirSim\MavLink
Com\MavLinkTest\MavLinkTest.vcxproj]

What should I do?

Zhao1ming, 2017-02-20:
"D:\AirSim\Air\AirSim\AirSim.sln" (default target) (1) ->
"D:\AirSim\Air\AirSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
"D:\AirSim\Air\AirSim\AirLib\AirLib.vcxproj" (default target) (3) ->
"D:\AirSim\Air\AirSim\MavLinkCom\MavLinkCom.vcxproj" (default target) (4) ->
(ClCompile target) ->
 
 d:\airsim\air\airsim\mavlinkcom\src\serial_com\tcpserver.hpp(8): fatal error C1083: Cannot open include file: 'boost/
array.hpp': No such file or directory [D:\AirSim\Air\AirSim\MavLinkCom\MavLinkCom.vcxproj]
 
 d:\airsim\air\airsim\mavlinkcom\src\serial_com\udpserver.hpp(8): fatal error C1083: Cannot open include file: 'boost/
array.hpp': No such file or directory [D:\AirSim\Air\AirSim\MavLinkCom\MavLinkCom.vcxproj]

  d:\airsim\air\airsim\mavlinkcom\src\serial_com\udpclientport.hpp(8): fatal error C1083: Cannot open include file: 'bo
ost/array.hpp': No such file or directory [D:\AirSim\Air\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  

d:\airsim\air\airsim\mavlinkcom\src\serial_com\tcpclientport.hpp(8): fatal error C1083: Cannot open include file: 'bo
ost/array.hpp': No such file or directory [D:\AirSim\Air\AirSim\MavLinkCom\MavLinkCom.vcxproj]

  d:\airsim\air\airsim\mavlinkcom\src\serial_com\serialport.hpp(7): fatal error C1083: Cannot open include file: 'boost/asio.hpp': No such file or directory [D:\AirSim\Air\AirSim\MavLinkCom\MavLinkCom.vcxproj]

  d:\airsim\air\airsim\mavlinkcom\src\serial_com\tcpserver.hpp(8): fatal error C1083: Cannot open include file: 'boost/array.hpp': No such file or directory [D:\AirSim\Air\AirSim\MavLinkCom\MavLinkCom.vcxproj]

  d:\airsim\air\airsim\mavlinkcom\src\serial_com\tcpserver.hpp(8): fatal error C1083: Cannot open include file: 'boost/array.hpp': No such file or directory [D:\AirSim\Air\AirSim\MavLinkCom\MavLinkCom.vcxproj]

  d:\airsim\air\airsim\mavlinkcom\src\serial_com\serialport.hpp(7): fatal error C1083: Cannot open include file: 'boost/asio.hpp': No such file or directory [D:\AirSim\Air\AirSim\MavLinkCom\MavLinkCom.vcxproj]

  d:\airsim\air\airsim\mavlinkcom\src\mavlinksemaphore.cpp(6): fatal error C1083: Cannot open include file: 'boost/date_time/posix_time/posix_time.hpp': No such file or directory[D:\AirSim\Air\AirSim\MavLinkCom\MavLinkCom.vcxproj]

  d:\airsim\air\airsim\mavlinkcom\src\serial_com\tcpserver.hpp(8): fatal error C1083: Cannot open include file: 'boost/array.hpp': No such file or directory [D:\AirSim\Air\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  
d:\airsim\air\airsim\mavlinkcom\src\serial_com\tcpserver.hpp(8): fatal error C1083: Cannot open include file: 'boost/array.hpp': No such file or directory [D:\AirSim\Air\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  
d:\airsim\air\airsim\mavlinkcom\src\impl\mavlinkftpclientimpl.cpp(5): fatal error C1083: Cannot open include file: 'b
oost/algorithm/string.hpp': No such file or directory [D:\AirSim\Air\AirSim\MavLinkCom\MavLinkCom.vcxproj]

Also some other faults, it looks like the problem of boost.

lovettchris, 2017-02-21:
Did you follow the instructions to [Install Boost](https://github.com/Microsoft/AirSim/blob/master/docs/install_boost.md) ?  And set the BOOST_ROOT environment variable?

Zhao1ming, 2017-02-21:
Yes, I think I should do this:

The following directory should be added to compiler include paths:

    D:\AirSim\boost\boost_1_63_0\boost_1_63_0

The following directory should be added to linker library paths:

    D:\AirSim\boost\boost_1_63_0\boost_1_63_0\stage\lib

But how?

Zhao1ming, 2017-02-21:
Amazing! I repair VS 2015 and restart my computer, now it works. Thanks!

lovettchris, 2017-02-22:
Oh, yes, when you modify your system environment variables, you need to start whatever process you want to pick up those new variables.  So restarting VS picked up BOOST_ROOT.

closed 2017-02-22 by lovettchris