microsoft/AirSim #40: Unable to connect to SITL

DanWilson00, 2017-02-21:
I have successfully built and run PX4 SITL in Windows 10 Bash and get:
`INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..`
TCPView confirms that PX4 is connected to localhost and port 14560

I have then done the following:
- Downloaded/installed UE4
- Downloaded/built the dependencies
- Downloaded/built Airsim in VS2015
- Created a new project, added an environment, copied the Airsim plugin files to the project, enabled the plugin
- Modified settings.json to not use serial
- Ran the UE4 project in VS2015 and changed to SimGameMode
- Pressed play

The simulator doesn't connect to the SITL and UE4 hangs until I stop running the project from VS2015 (see below)

[image]

Am I missing something?

sirluzelot, 2017-02-21:
Modify LocalHostIp to something different than 127.0.0.1, for example 127.0.0.2, that worked for me. Matching IPs makes it use connectLocalUdp which actually starts a server instead of listening, therefore the bind error if the PX4 firmware is already running

DanWilson00, 2017-02-21:
That worked, thanks sirluzelot!

closed 2017-02-21 by DanWilson00