microsoft/AirSim #41: Arming Denied. Preflight checks have failed

SeowooJang, 2017-02-22:
I had followed the instruction and was able to load environments with the drone from the AirSim. But when I hit the play button in unreal, the meesage "Arming Denied. Preflight checks have failed" in green shows up and I cannot control the drone at all with rc or droneshell. Also, the three images (depth, segmentation and rgb images) show only white images.

Can someone help me to figure the problem out?

lovettchris, 2017-02-22:
Try setting your PX4 to Simulation mode using QGroundControl by selecting this AirFrame:

[cid:image001.png@01D28C79.983EA720]

SeowooJang, 2017-02-22:
I cannot see the image you have posted.

In Airframe, I had selected Simultion --> HIL Quadcopter X 
I am using different remote controller from the recommended one, though.

Can controller settings be a reason for the white images as well?

Thank you for answering.

lovettchris, 2017-02-22:
It is odd that the drone goes straight into arming.  Did you checkout my [Walkthrough Demo Video](https://www.youtube.com/watch?v=HNWdYrtw3f0).

SeowooJang, 2017-02-22:
I found that the "Arming Denied" issue was due to my pixhawk hardware and fixed it. 
However, still unable to control drone through RC or droneshell along with the white output images.

sytelus, 2017-02-22:
Please post error messages you see. You might want to make sure your RC is calibrated and QGroundControl is showing everything green. Please review the video @lovettchris has mentioned above.

SeowooJang, 2017-02-22:
This is the screen that I get. 

[image]

lovettchris, 2017-02-22:
Hmm, that is unusual.  I wonder if your start point is slightly under ground such that the camera is recording nothing under your world?  I see freefall detected, so looks like your drone was above the world then came down and landed, how far above was it?  I usually like to set up the PlayerStart so that the drone drops only a few inches onto a smooth takeoff area.

SeowooJang, 2017-02-22:
It was because of the version of my Unreal Engine. 
I unintentionally installed 4.15 in the first place which resulted in the white images.
Using 4.14.3 produces the depth, segmented and rgb images well.

Thank you for the help!
I appreciate it.

closed 2017-02-22 by SeowooJang