microsoft/AirSim #42: SITL: connection setup is incorrect

bkueng, 2017-02-22:
AirSim cannot not connect to SITL after adjusting the settings accordingly. It throws an error that the port is already in use. To make it work, I had to change https://github.com/Microsoft/AirSim/blob/master/MavLinkCom/src/impl/MavLinkConnectionImpl.cpp#L58 from the local to the remote port (basically using `connectRemoteUdp` instead). This is then correct because the PX4 simulator specifies its local port on startup.

I'm on Linux, but it seems to be the same problem on Windows: https://github.com/Microsoft/AirSim/issues/40

lovettchris, 2017-02-22:
Are you sure you are on the latest bits?  The simulator should not be calling connectLocalUdp, it should be calling connectRemoteUdp when you change the settings such that UdpIp is a remote Ip and UseSerial is false.  Specifically take a look at the if-test in [createHILUdpConnection](https://github.com/Microsoft/AirSim/blob/master/AirLib/src/control/MavLinkHelper.cpp#L288).

bkueng, 2017-02-22:
Yes, I just tested current master. It does call `connectLocalUdp` because everything runs on the same machine (no VMs involved). Configured IP:
```
    "UdpIp": "127.0.0.1",
```

clovett, 2017-02-22:
I see so are you saying we should remove that if test and always use connect remote?

bkueng, 2017-02-22:
If that's the only use-case then I think yes, we can remove it.

lovettchris, 2017-02-23:
[Fixed](https://github.com/Microsoft/AirSim/commit/46b00c2335ace6efdd965af718bb6725e747a51f)

closed 2017-02-23 by lovettchris