microsoft/AirSim #44: Can a stereoscopic camera be enabled?

t4nuj, 2017-02-22:
Is it possible to collect stereoscopic images as well as depth maps from the simulation?

sytelus, 2017-02-22:
Yes, we have multi-camera support in the code. I'm working on making it work up stream so this should be available in ~2 weeks.

robbel, 2017-02-25:
@sytelus: is there a way to spread the rendering of multiple camera views across a cluster of computers? Say we'd like to add 8 cameras -- I assume a single computer may not be able to render all those?

sytelus, 2017-03-01:
Each camera render has about 10-20% hit on FPS so running with > 3 cameras would lower FPS too much. There may be few ways to achieve this (1) use multiplayer features to run same environment on many machines (2) use time dilation (3) move same camera around 8 times. We have no plans for (1) yet but we welcome anyone if they want to contribute that feature. The (2) is in plans but on lower priority for now although current code should naturally support it. The (3) is tricky but more easily doable.

I would be interested why you might need so many cameras, of course :).

robbel, 2017-03-05:
@sytelus, thanks a lot for the info. The multi-camera topic mostly concerned autonomous ground vehicle simulation, where 7+ cameras can be quite likely.

Strupi11, 2017-06-13:
@sytelus Whats the status here? I would also need the camera viewports as RTSP streams. I could also help finishing it, if you'd push your branch.

sytelus, 2017-07-19:
This is implemented now. Please the doc https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md. I have also wrote sample code in Examples project that retrieves stereo images and ground truth depth image.

closed 2017-07-19 by sytelus