microsoft/AirSim #47: How to add AirSim plugin to Unreal project

cborelc, 2017-02-22:
Hi:
I succeeded (or at least I think I did) building the AirSim plugin and it created a folder Airsim in "Plugins" with 2 sub-folders: "Content" and "Source" and a "AirSim.uplugin" file. There are no DLL files in these directories - is that correct?

After I copy the "Plugins" folder to a freshly created project with no "starter" content and adding the AirSim entries to the uproject file I try to run "generate visual studio project files" and I get an error: 

"This project does not have any source code. You need to add C++ source files to the project from the Editor before you can generate project files."

No I am stuck - I can't even open the uproject file in the Unreal editor since I get an error:

"The following modules are missing or built with different engine version:
UE4Editor-Myproject2.dll
UE4Editor-Airsim.dll"
Would you like to rebuild them now?"

If I say "Yes" the I get the message:

"MyProject2 could not be compiled. Try rebuilding from source manually."

But there is no sln file in source - only in AirLib and that one does not build when I click on it.

What am I doing wrong here?
Thanks
Chris

This is how my uproject file looks like:
{
	"FileVersion": 3,
	"EngineAssociation": "4.14",
	"Category": "",
	"Description": "",
"Modules": [
            {
                "Name": "MyProject2",
                "Type": "Runtime",
                "LoadingPhase": "Default",
                "AdditionalDependencies": [
                    "AirSim"
                ]
            }
        ],
	"Plugins": [
		{
			"Name": "Substance",
			"Enabled": true,
			"MarketplaceURL": "com.epicgames.launcher://ue/marketplace/content/2f6439c2f9584f49809d9b13b16c2ba4"
		},
{
                "Name": "AirSim",
                "Enabled": true
            }
	]
}

fmzhu, 2017-02-22:
hi, I've got the same problem, I'm stuck at it!

camow7, 2017-02-22:
I found that you actually need starter content to get it going. You may able to do it without if your familiar without UE4 but if not best option is to tick the starter content box. Did you successfully execute the “build” command? Are you running all your commands inside the VS2015 command terminal? Have you added boost to your environment variable?

lovettchris, 2017-02-22:
The fix for the error "This project does not have any source code. You need to add C++ source files to the project from the Editor before you can generate project files."

is to use the Unreal Editor to Create New Project, select the C++ tab, rather than the Blueprint tab, and I normally start with no starter content, then drop the content in from a downloaded environment, like LandscapeMountains using "Add to Project".  Then edit the .uproject to include the AirSim plugin.

You may also need to add or edit the PlayerStart object so that it is slightly above ground near a good space where the drone can take off.  Then hit play, and the plugin should add the drone and you should be good to go.

fmzhu, 2017-02-22:
@lovettchris thank you for your help, so which kind of C++ template should I use, the "Basic Code" one?

cborelc, 2017-02-22:
Thanks for responding so fast - I am sorry to be such a NOOB (not being a VS person at all):

The fix for the error "This project does not have any source code. You need to add C++ source files to the project from the Editor before you can generate project files."

Q: By the "editor" do you mean VS2015 (not the UE4 editor) and the location is the "Solution explorer"
I did that and it actually built without errors and now I have 3 DLL's . Now is it the time to make the changes in the uproject file?

cborelc, 2017-02-22:
After adding the lines in uproject I am not able to open the file - I get a message: 

"Unable to read project status."

fmzhu, 2017-02-22:
@cborelc what is your UE-version ? 4.15 ?

cborelc, 2017-02-23:
UE 4.14.3 as prescribed. It would be nice to have a video showing how to go from adding the airsim plugin to a simple UE4 project (not some downloaded scene which most people don't have). I am new to UE4 and many things are not clear to me.

To answer the other questions by camow7:

Did you successfully execute the “build” command? 
>>yes
Are you running all your commands inside the VS2015 command terminal? 
>> yes
Have you added boost to your environment variable?
>> yes

sytelus, 2017-02-25:
The error "This project does not have any source code." usually occurs if you started Blueprint project and dropped AirSim plugin in to it. Unreal has two types of projects: C++ and Blueprint. The AirSim plugin works with C++ projects (it can also work with Blueprint project if you had .cpp file in your project). If you had created C++ project and still see this error then try adding a C++ file in your project by going to Unreal Editor and go to File > New C++ class. Just enter some values, leave everything blank. Close editor, copy AirSim to Plugins folder, right click on uproject and generate VS project files. I have never seen this not working with Unreal 4.14.

Detailed steps to create Unreal project are [here](https://github.com/Microsoft/AirSim/blob/master/docs/build.md).

@lovettchris also has [uploaded video](https://www.youtube.com/watch?v=HNWdYrtw3f0&feature=youtu.be) (see starting at 3:50).

Zhao1ming, 2017-02-26:
@lovettchris Hi, can you explain it more specific? How to "drop the content in from a downloaded environment, like LandscapeMountains using "Add to Project", actually I can not find "Add to Project" button. Thanks

cborelc, 2017-02-27:
I followed what @sytelus suggested and - wow it worked! I did not have to add the lines uproject and I did not get errors when opening the uproject by double click it. So I finally got the game running in SimGameMode to work and display the depth, object class and rendered image from the drone:
[image]
 Now I think I can get further with this project. Thanks very much!

Zhao1ming, 2017-02-27:
That's cool!  But can you tell how to **add  .cpp file into  Blueprint project**  with more detailed? A lot thanks.

cborelc, 2017-02-27:
How to build a AirSim project in UE4:

1. Lauch UE4 editor 4.14.3

2. Select new project, C++ "Basic code" project with Starter content (or your content - you can add content later if you wish)

3. Wait and this starts VS2015 and later UE4

4. In Windows workspace copy plugins/SimAir to project folder

5. In VS2015 go to "Solutions Explorer" and right click on "MyProject" and select "add "Existing item" go in file selector and select one of the CPP files in AirSim, e.g. "AirBluePrintLib.cpp"

6. Build project (this might not be necessary but I did it just to test if the plugin was actually added)

my output at this stage:
1>------ Build started: Project: UE4, Configuration: BuiltWithUnrealBuildTool Win32 ------
2>------ Build started: Project: MyProject4, Configuration: Development_Editor x64 ------
2>  Performing full C++ include scan (hot reloading a new target)
2>  Creating makefile for hot reloading MyProject4Editor (no existing makefile)
2>  Compiling game modules for hot reload
2>  Parsing headers for MyProject4Editor
2>    Running UnrealHeaderTool "D:\Borel\Unreal\MyProject4\MyProject4.uproject" "D:\Borel\Unreal\MyProject4\Intermediate\Build\Win64\MyProject4Editor\Development\MyProject4Editor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
2>  Reflection code generated for MyProject4Editor in 4.7186912 seconds
2>  Performing 5 actions (4 in parallel)
2>  [3/5] Resource PCLaunch.rc
2>  [4/5] Resource ModuleVersionResource.rc.inl
2>  Module.AirSim.cpp
2>  AirSim.generated.cpp
2>D:\Borel\Unreal\MyProject4\Plugins\AirSim\Source\MavMultiRotor.cpp(149): warning C4458: declaration of 'total_' hides class member
2>  d:\borel\unreal\myproject4\plugins\airsim\source\MavMultiRotor.h(67): note: see declaration of 'MavMultiRotor::total_'
2>  [5/5] Link UE4Editor-AirSim-3159.dll
2>  AirLib.lib(Settings.obj) : MSIL .netmodule or module compiled with /GL found; restarting link with /LTCG; add /LTCG to the link command line to improve linker performance
2>     Creating library D:\Borel\Unreal\MyProject4\Plugins\AirSim\Intermediate\Build\Win64\UE4Editor\Development\UE4Editor-AirSim-3159.lib and object D:\Borel\Unreal\MyProject4\Plugins\AirSim\Intermediate\Build\Win64\UE4Editor\Development\UE4Editor-AirSim-3159.exp
2>  Generating code
2>  Finished generating code
2>  Total build time: 102.17 seconds
========== Build: 2 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========

7. In VS2015 select "save all" to save  project, close UE4 editor

8. Start UE4 editor by double clicking Myproject.uproject file

9. Build project (again) after an error message says it was built for a previous version (This is probably because it was built before in VS2015 as a test)

10. UE4 starts up and you should see the table and chairs of the starter scene and there should be no further errors.

11. Now change the game mode by selecting:
Blueprints -> Project settings: Gamemode -> Select GameModeBase Class -> SimGameMode
and
Blueprints -> World override: Gamemode -> Select GameModeBase Class -> SimGameMode

12. Select "Play" and it should look like the image below
[image]

cborelc, 2017-02-27:
As far as this problem I am happy with the result and consider the issue closed.

closed 2017-02-27 by cborelc