microsoft/AirSim #48: Rpc control

jamuus, 2017-02-22:
I can't seem to get RPC control of the quad. I'm using the SITL method and have gotten manual control working using QGC. The problem I seem to be having is getting the controller into offboard mode; I'm not sure how AirLib handles putting the quad into this mode.

DroneShell times out when performing requestcontrol and when attempting to change in QGC I get "command temporarily rejected".

Anyone have any pointers for debugging this?

sytelus, 2017-02-25:
I'd seen this error before but can't recall the context yet. Can you confirm if your basic setup works. That is,

1. You have Unreal project working with SITL.
2. You are able to arm and takeoff using QGC.
3. You shutdown QGC, start DroneShell, type Arm and then TakeOff.

See video walkthrough starting here: https://youtu.be/HNWdYrtw3f0?t=506

lovettchris, 2017-02-27:
And for moveTo* methods you need to use "requestControl" on the DroneShell command line.  Is that the part that is failing?

jamuus, 2017-02-27:
I've followed the steps above but requestControl is still hanging/timing out. I've managed to get control using dronekit's example px4 script but this seems to put the controller in AUTO mode rather than OFFBOARD.

Timebutt, 2017-03-06:
Hi @jamuus, are you getting the same behaviour as me? See issue #85. I have the feeling we are experiencing the same problem.

jamuus, 2017-03-06:
Yeah I expect I am. I haven't managed to get OFFBOARD mode working using SITL.

closed 2017-03-26 by jamuus