microsoft/AirSim #50: How can i see the three cameras on the unreal environment hud?

FRopero, 2017-02-23:
Hi!,

I'm trying to attach the three cameras of the UAV in my Unreal environment, but i don't get it. I can see the UAV and it is running under the SITL mode perfectly.

Thanks!

sytelus, 2017-02-25:
I just created doc just for you :).

Once you see plugin contents, go to Blueprints and open the BP_FlyingPawn. There are two PIPCamera on the its body.

closed 2017-02-25 by sytelus

FRopero, 2017-02-25:
Thanks, much appreciate! Running

Cheers!

nacho-carnicero, 2017-05-10:
Hi, 
I'm facing a similar problem. I've built AirSim and everything works fine with the PX4 SITL control laws, but I can't manage to see the depth, segmentation and FPV cameras. As soon as I hit play there are no cameras showing on the lower part of the screen, and if I hit "[" or "]" to try to switch to the first person view camera the simulator doesn't do anything (could this be due to the fact that I have to press "Alt Gr" to get "[" or "]" ? I have a spanish keyboard)

Cheers, 

Nacho.

lovettchris, 2017-05-10:
Try clicking on the view to make sure it has keyboard focus.  The camera views are available with keystrokes "1", "2", or "3" or type "0" to get all of them.

nacho-carnicero, 2017-05-11:
Thanks for the reply @lovettchris, yesterday I wasn't able to access the cameras with these keystrokes, even with keyboard focus on unreal engine (I was able to hit F1 and access the small yellow menu).
Today I'm building it on Linux (nothing to do with the camera issue) so I'll let you know if I'm able to get the cameras.

nacho-carnicero, 2017-05-19:
Sorry I forgot to update, finally it worked fine. I was just pressing 0 in the numpad instead of the regular numbers. Nevertheless I'm still not able to get the FPV view, since the keystroke is [ or ] and in a spanish keyboard you have to press ALT Gr to get that key. Don't you think it would be more convenient to assign a keyboard letter instead of a sign @lovettchris such as for toggling the debug report which uses "R"?

christian33moreno, 2022-08-13:
> I just created [doc ](https://github.com/Microsoft/AirSim/blob/master/docs/working_with_plugin_contents.md) just for you :).
> 
> 
> 
> Once you see plugin contents, go to Blueprints and open the BP_FlyingPawn. There are two PIPCamera on the its body.

Once we have located the BP_PIPCameras, how do we actually get them to appear on the screen?

sandilyasg, 2023-11-04:
How do I use Airsim HUD with multiple quadrotors running? I want to visualize RGB, Depth, and Semantic label frame for each drone. The default version shows me the 3 cameras only but I am looking to expand to multiple quadrotors.