microsoft/AirSim #51: How to Use Programmatic control?

zhongtao93, 2017-02-24:
Is there a usage in more details about how to use programmatic control?
I build HelloDrone and run it, A shell appears and  shows that "Press Enter to arm the drone ", then how to connect the Unreal Project. 
 After reading https://github.com/Microsoft/AirSim/blob/master/docs/apis.md I still feel confused to use it.

Thank you!

lovettchris, 2017-02-25:
We have some limitations to watch out for that we are working on improving, but for now:
1) The Unreal Project needs to be running (Click Play) before you try and connect the HelloDrone app.
2) There is a bug such that if you "Stop" the unreal project you have to stop the UE Editor and restart it, otherwise the server will not reconnect to the Drone client api.  

These are rpclib limitations/bugs and we plan to fix them, just don't have it in yet...  

If you want to dig deeper the RpcLibServer is running in the Unreal project if you have added the Plugin correctly.  Then any other process can connect to that server using RpcLibClient.  These communicate over TCP.  Note that both RpcLibClient and RpcLibServer can also take ip addresses, and this enables RPC over the network instead of on the same machine.  But of course if you do it over the network you need to also worry about any firewalls, so you may need to open the default RpcLibClient port of 41451.

sytelus, 2017-02-25:
Hi @zhongtao93, just to add on...

@lovettchris had posted video on how to use DroneShell. Here's the segment that goes over DroneShell: https://youtu.be/HNWdYrtw3f0?t=510

DroneShell is just more featureful version of HelloDrone. As Chris mentioned above, first run Unreal project. Make sure you are able to take off drone manually with RC and fly it around so you know everything is working. Then stop editor and start again. Run HelloDrone and execute commands there. Currently we have commands like takeoff, land, moveToPosition, moveByVelocity working. For other commands we have found one bug that we are working on fix in coming few days.

zhongtao93, 2017-02-26:
Thank you for your answer!
Hi, @sytelus, do I must have a RC to connect it before programmatic control? I run Unreal project and it shows as below:
[image]

Thank you.

sytelus, 2017-02-26:
You do need PX4 in hardware-in-loop or software-in-loop. One other problem is that PX4 might not allow you to fly if you don't have RC because in real life it would be unsafe because there is no possibility of manual override or kill switch. I haven't tested this so just my guess.

In next week or two, we plan to release Standalone Mode which will allow you to use AirSim without PX4. In this mode, you should be able to use API without RC or any additional harddware. However XBox or other game pad might still be required for manual flight.

zhongtao93, 2017-02-26:
OK, thank you very much. I'm look forward to the next release

closed 2017-03-01 by sytelus

udaydlsv, 2017-04-03:
Do we have the standalone mode yet ? I do not have any additional hardware and tried to control the drone with droneshell. I get the following error. Is this expected ? 
[image]

lovettchris, 2017-04-03:
standalone is still in development.  But you can use [sitl mode](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md) without any additional hardware.

pdebitetto-tft, 2017-12-13:
Hi Chris, the link to SITL mode does not work.  did the functionality go away?  
Thanks
Paul

delphifissure, 2018-06-05:
Here's a +1 for SITL mode, is that gone? Would love to be able to test programmatic flight control with physics before I pull the trigger on drone hardware.

Thanks!

vinbo, 2018-06-26:
I found it too...