microsoft/AirSim #53: Is there any plan for ROS integration like Gazebo does?

JiamingSuen, 2017-02-25:
Testing control algorithms with [SITL plugin](https://github.com/erlerobot/ardupilot_sitl_gazebo_plugin) + Gazebo + ROS is now the common way, while Gazebo only provides poor visual simulations which is not realistic enough for Deep Learning/Reinforcement Learning research. Is there any plan for AirSim of such kind of integration with ROS? In specific this would allow us to retrieve state information and camera view etc from AirSim, and we can use tools like [mavros](http://wiki.ros.org/mavros) to send control command to the vehicle.
Thanks for this incredible work!

sytelus, 2017-02-25:
Yes, we do have plan to enable ROS for AirSim. We have some code that was working with Matrice underROS which we will port to AirSim. This will essentially make possible to call our APIs as ROS services and request publication of sensor data.

I'll change the title of this issue to reflect this and keep this open.

AlexandreBorowczyk, 2017-03-28:
Have anyone tried to integrate SITL to AirSim?

lovettchris, 2017-03-28:
Sure, see [https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md).

zaddan, 2017-06-22:
Hi,
Is there any update on the ROS integration?

sytelus, 2017-07-19:
Not yet. We are heads down on making Linux build fully work first and create standalone flight controller for next two weeks or so. After that ROS integration would take priority.

togaen, 2017-09-30:
@sytelus Are there any examples or readme's you could point to that show the kind of AirSim ROS adapter you mentioned in #333?

sytelus, 2017-10-06:
Not yet. Basically the plan is to use RpcApi to transmit ROS messages.

jokla, 2017-10-19:
Hi @sytelus ! Do you have any update on the ROS integration?
Thank you!

pdebitetto-tft, 2017-12-13:
Hi Guys, just checking back about ROS integration.  I'm just learning about AirSim and compared to the others this has many desirable characteristics.  Really nice work Chris et. al.!  however i would like to drive it using a higher level mission-autonomy computer running ROS.  

Also, i would like to run a SITL version using the PX4 stack.  the sitl.md file recommended above doesn't seem to exist anymore.   
Thanks!
Paul

pdebitetto-tft, 2017-12-13:
Hi Guys, just checking back about ROS integration. I'm just learning about AirSim and compared to the others this has many desirable characteristics. Really nice work Chris et. al.! however i would like to drive it using a higher level mission-autonomy computer running ROS.

Also, i would like to run a SITL version using the PX4 stack. the sitl.md file recommended above doesn't seem to exist anymore.  If there are anymore recent developments please let know.

Thanks!
Paul

HarrisSym, 2018-01-31:
Hello, i'm intestred in ROS integration. Is there any update on this matter?

ykhedar, 2018-04-03:
Hi,This would be a great addition to the simulation environment. Is the pull request number 842 **rosbridge_first_test** the official addition of this feature or a different version being worked on by the AirSim team. Would be great to know the status. Really appreciate the great work being done by the AirSim team :)

Thanks,
ykhedar

sytelus, 2018-04-11:
rosbridge_first_test is by one of our users and is under review. Please feel free to review/provider feedback for this PR.

n1ckfg, 2018-04-25:
We're making progress on the bridge. With the new API, is the car coordinate system still NED, or had that changed?

sytelus, 2018-04-25:
Everything is still NED.

MohamedAliRashad, 2018-05-13:
Waiting for the ROS integration as well.
 
This simulator looks Damn Good guys, Great Work :+1:

wndaiga, 2018-06-06:
Very promising work! This would easily blow my other options for a quad simulator out the water (TUM simulator, Hector Quadcoter) but the lacking ROS/MoveIt! integration is my only hold out

TritonSailor, 2018-06-06:
You can find some simple examples I contributed yesterday for publishing AirSim data in ROS nodes using PythonClient in my PR here:

https://github.com/Microsoft/AirSim/pull/1135

Please try them and let me know if you have any issues running them.  I will be contributing additional examples in the near future. 

Sounds like a complete ROS integration solution is in the works, but in the meantime, these examples should help you get started.

wndaiga, 2018-06-09:
@TritonSailor Thanks a ton, exactly what I needed!

madratman, 2019-05-13:
Please see [airsim_ros_pkgs](https://github.com/microsoft/AirSim/tree/master/ros/src/airsim_ros_pkgs) and [airsim_tutorial_pkgs](https://github.com/microsoft/AirSim/tree/master/ros/src/airsim_tutorial_pkgs) for a ros wrapper over the multirotor API

closed 2019-05-13 by madratman

xlong0513, 2019-11-27:
Interested in ros and mavros integration. Is there any progress yet?