microsoft/AirSim #54: Connection to Drone failed, please check your setting .json

zf223669, 2017-02-26:
I have setup all the environment for the AirSim, However, when i run the Unreal Engine, some error showed at the left of the scene view, like this:

Connection to drone failed, please check your settings.json
Could not open the serial port COM3, error = -1
Connection to LogViewer failed, please check your settings.json
Connection to QGroundControl failed, please check your settings.json

I have modify the Local Host Ip/LogViewerHostIp\QgcHostIp to my local ip:the whole settings is :+1: 

{
  "AirControlCompID": 1,
  "AirControlSysID": 134,
  "ExtRendererCompID": 1,
  "ExtRendererSysID": 167,
  "ExternalSimPort": 14588,
  "LocalHostIp": "10.1.9.10",
  "LogViewerHostIp": "10.1.9.10",
  "LogViewerPort": 14388,
  "Pixhawk": {
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true
  },
  "QgcHostIp": "10.1.9.10",
  "QgcPort": 14550,
  "SimCompID": 42,
  "SimSysID": 142
}
Unfortunaly, it does not work either.......

sytelus, 2017-02-26:
Do you have PX4 connected to your PC via USB port (above settings says that you are trying to use PX4 via serial port)?

zf223669, 2017-02-26:
yep, i have connect the px4 to my pc. i use a xbox joystick connected to my pc and a PX4, have made all the setting from the QGroundController. But it do not work.

zf223669, 2017-02-26:
because i have no fr-sky radio receiver, i just connectted the PX4 to the PC by the USB port, and connected a xbox joystick to PC by the USB port also. Is that right?

lovettchris, 2017-02-27:
Ah, we don't yet support xbox controller as a way to fly the PX4.  But it is on our list of things to get to soon.

zf223669, 2017-02-27:
I have found the problem. Because i forgot to close the QGroundController when the unreal project start up.

lovettchris, 2017-02-27:
Ah, yes, that would do it.  You can only have one app connected to the serial port at a time.

closed 2017-02-27 by lovettchris