microsoft/AirSim #56: compiler error not found rpc.lib?

mulysatest, 2017-02-26:
Everything seems to be compile properly except this error 
LINK : fatal error LNK1181: cannot open input file 'rpc.lib'
I have search all of the folder and didn't see any rpc.lib at all

mulysatest, 2017-02-26:
The above error I got when I tried to compile to src with .sln file.
Then I tried again with buid.cmd instead. However it got stucked at this point
Note: current eigen path is c:\airsim\airsim\eigen3\eigen

Build FAILED.

"c:\airsim\AirSim\AirSim.sln" (default target) (1) ->
"c:\airsim\AirSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
"c:\airsim\AirSim\AirLib\AirLib.vcxproj" (default target) (3) ->
(ClCompile target) ->
  c:\airsim\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083:
 Cannot open include file: 'Eigen/Dense': No such file or directory [c:\airsim\
AirSim\AirLib\AirLib.vcxproj]
  c:\airsim\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083:
 Cannot open include file: 'Eigen/Dense': No such file or directory [c:\airsim\
AirSim\AirLib\AirLib.vcxproj]
  c:\airsim\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083:
 Cannot open include file: 'Eigen/Dense': No such file or directory [c:\airsim\
AirSim\AirLib\AirLib.vcxproj]
  c:\airsim\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083:
 Cannot open include file: 'Eigen/Dense': No such file or directory [c:\airsim\
AirSim\AirLib\AirLib.vcxproj]
  c:\airsim\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083:
 Cannot open include file: 'Eigen/Dense': No such file or directory [c:\airsim\
AirSim\AirLib\AirLib.vcxproj]
  c:\airsim\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083:

mulysatest, 2017-02-26:
I managed to fix problem by move c:\airsim\airsim\eigen3\eigen\eigen to C:\airsim\AirSim\AirLib\include\eigen

lovettchris, 2017-02-27:
You shouldn't have to move Eigen.  I have my eigen directory on a different drive and it works fine.  So I'm guessing your EIGEN_ROOT environment variable was not set correctly.  See [install eigen](https://github.com/Microsoft/AirSim/blob/master/docs/install_eigen.md).

mulysatest, 2017-02-27:
Yes I read the installation manual many times but it failed. Frankly speaking the installation manual missing many steps and lack to detail instruction especially it need to mentio about cmake as the build script need it.

lovettchris, 2017-02-27:
You are correct, the cmake dependency is new, we added that in order to build "rpclib" instead of shipping rpclib binaries.  This solved some cross-platform build issues.  I added the following additional step to the build steps:

3. Install [cmake](https://cmake.org/download/) which is used to build the rpclib.

Please continue to report any other build errors that you run into, thanks.  I also posted a new [Unreal AirSim Setup  Video](https://youtu.be/1oY8Qu5maQQ) that shows you all the setup steps needed to get the plugin working in a brand new unreal project.

lovettchris, 2017-03-01:
Are you on Windows ?  Try running "git submodule init" and "git submodule update", you will see this in the "build.cmd" script.

jjy201314, 2017-03-05:
how to install the cmake and build the rpclib? please

jjy201314, 2017-03-05:
D:\wurenji\AirSim-master>build.cmd
CMake Error: The source directory "D:/wurenji/AirSim-master/external/rpclib" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
Error: could not load cache
Error: could not load cache
#### Build failed

how to resolve this problem?

lovettchris, 2017-03-05:
What do you see in that folder?  If it is empty, try running "git submodule init" and "git submodule update" from the AirSim root folder.  This should pull in the external/rpclib folder, and if that succeeds there definitely should be a CmakeLists.txt in there.  You could also try "git submodule update -f" from the AirSim root folder to force the submodule to be updated.

lovettchris, 2017-04-05:
closing due to inactivity.

closed 2017-04-05 by lovettchris

jonweili, 2017-09-20:
Hello, i am confused about building airsim. After running build.cmd,it seems successful,however, there is nothing changed in fold Unreal\Plugins, no more plugin bits exist in this fold. what is wrong about this?
Expected your answer, thanks!@lovettchris

tanmoy-sabir, 2017-11-13:
I am facing these problem while compiling build.cmd. 
I have no idea how to solve this problem.

  DroneControllerBase.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
  MultirotorRpcLibServer.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
  MultirotorRpcLibClient.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
  CarRpcLibServer.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
  CarRpcLibClient.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
  SafetyEval.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
  ObstacleMap.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
  RpcLibServerBase.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
  RpcLibClientBase.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]

coldwarrior5, 2019-02-08:
Hi guys, I found what is the issue. The rpc.lib is built for x64, not x86 and when you try to build the project using default settings it fails.
To have a successful build you need to change platform to x64 you can find the setting between ReDo button and Local Windows Debugger.
Being Debug or Release doesn't matter of course because it was built for both.

stavBodik, 2019-04-17:
> Hi guys, I found what is the issue. The rpc.lib is built for x64, not x86 and when you try to build the project using default settings it fails.
> To have a successful build you need to change [...]

@coldwarrior5 
Good Job