microsoft/AirSim #57: global shader cache file '.PCD3D_SM5.bin.' is missing

julianoes, 2017-02-26:
Exciting project, thanks for making this available and the detailed instructions!

I'm at the point where I can almost compile my (empty) project. However, I get the message:

```
The global shader cache file 'C:/Program Files/Epic Games/UE_4.15/Engine/GlobalShaderCache-PCD3D_SM5.bin' is missing. 
```

[image]

It is correct that the file does not exist. Is there a setting or a step I'm missing? Thanks.

lovettchris, 2017-02-28:
Try the "DebugGame Editor" build configuration.  Looks like you are trying to run the "shipping" bits, which require cooking the assets.  If you use "DebugGame Editor" or "Development Editor" it will be easier to just press F5, which launches the UE editor, the "Play" to play the game.  These modes are slower to load because the content is not "cooked", but as a result the "just work".  Cooking the assets can be done using File menu of UE editor.  For example, if you built "Shipping" config using VS, then set the same build configuration in UE editor, File/Package Project/Build Configuration.  Then select File/Package Project/Windows/64 bit.  This will give you a .exe that you can run and it will load much quicker.  See also [Unreal AirSim Setup Video](https://youtu.be/1oY8Qu5maQQ).

julianoes, 2017-03-08:
@lovettchris thanks for your answer.

I realized I was using Unreal Engine 4.15.0 instead of 4.14.3. I downloaded 4.14.3, and created a new project and this time, was able to compile and run it.

closed 2017-03-08 by julianoes

sytelus, 2017-09-19:
I got same error and it was because I chose (by mistake) "DebugGame" instead of "DebugGame_Editor". Unfortunately this is not easily visible in VS so you need to drop down menu and go to Solution Configuartion.

kiichigo, 2017-10-04:
There is a way to increase the width of the drop down menu.

https://docs.unrealengine.com/latest/INT/Programming/Development/VisualStudioSetup/index.html#recommendedsettings