microsoft/AirSim #59: Missing  UE4Editor-AirSim.dll?

mulysatest, 2017-02-27:
I have successfully compile the plugin, however once I edit the .uproject file in order to enable plugin using notepage. Thank I confused with this step: 

"Right click on the Unreal project .uproject file, then Generate Visual Studio Project files"

I have right click on the project file, but there is no option to generate visual studio project.
Then I tried to open it using Unreal Editor and I got following msg show up:

---------------------------
Missing MyProject2 Modules
---------------------------
The following modules are missing or built with a different engine version:

UE4Editor-AirSim.dll

Would you like to rebuild them now?
---------------------------
Yes   No   
---------------------------

So, my question is
1. Where is UE4Editor-AirSim.dll?
2. Using notepad to manually add the new section is proper way to do it? Is there anyway to do it using the Unreal Editor?

sytelus, 2017-02-27:
Looks like your Unreal engine installation has problems. How did you installed it? Do you have Visual Studio 2015 Update 3 installed?

Also see: https://answers.unrealengine.com/questions/464207/generate-visual-studio-project-option-missing.html

mulysatest, 2017-02-27:
Download from website and install it normally and everything work fine (except once tried to enable Airsim plugin). Is the right click menu do the same as result once I click "Yes"  button?  
Once I click it tried to compile the code it said something as shown in this pictures https://drive.google.com/drive/u/0/folders/0BxxG1gezRx53UVdMLUVVeWVJbWc

clovett, 2017-02-27:
You can also generate the visual studio project filled using the UE Editor, look for menu item named "refresh visual studio project"

mulysatest, 2017-02-27:
I think I can't even entering the UE interface I the alert msg block me.
I curious that it might be related to UE4Editor-AirSim.dll ? Do you have this file in your compiled Airsim's plugin folder?

mulysatest, 2017-02-27:
I managed to get right click menu back and then click the generate visual studio project file and got this error (I have check that my EIGEN_ROOT point to proper location)

Discovering modules, targets and source code for project...
ERROR: Unable to instantiate instance of 'AirSim' object type from compiled assembly 'MyProject2ModuleRules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.  Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements.  The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable.  See readme: https://github.com/Microsoft/AirSim/blob/master/docs/install_boost.md
   at AirSim.AddEigenDependency() in c:\Users\pariwat\Documents\Unreal Projects\MyProject2\Plugins\AirSim\Source\AirSim.Build.cs:line 78
   at AirSim..ctor(TargetInfo Target) in c:\Users\pariwat\Documents\Unreal Projects\MyProject2\Plugins\AirSim\Source\AirSim.Build.cs:line 65
   --- End of inner exception stack trace ---
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   at System.Reflection.RuntimeConstructorInfo.Invoke(BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at System.RuntimeType.CreateInstanceImpl(BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes, StackCrawlMark& stackMark)
   at System.Activator.CreateInstance(Type type, BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes)
   at System.Activator.CreateInstance(Type type, Object[] args)
   at UnrealBuildTool.RulesAssembly.CreateModuleRules(String ModuleName, TargetInfo Target, FileReference& ModuleFileName)

Timebutt, 2017-02-27:
I am having the exact same issue: UE4Editor-AirSim.dll is missing.
In my case, I guess it has to do with the fact that I can't get my code built (32 errors on building AirSim, trying to figure out what went wrong).

[image]

I'm watching this issue to see if I can help provide more information!

mulysatest, 2017-02-27:
@Timebutt  Does the right click on .uproject to generate VS project work for you? Do you got same error as below?

Discovering modules, targets and source code for project...
ERROR: Unable to instantiate instance of 'AirSim' object type from compiled assembly 'MyProject2ModuleRules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'. Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements. The CLR exception details may provide more information: System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable. See readme: https://github.com/Microsoft/AirSim/blob/master/docs/install_boost.md
at AirSim.AddEigenDependency() in c:\Users\pariwat\Documents\Unreal Projects\MyProject2\Plugins\AirSim\Source\AirSim.Build.cs:line 78
at AirSim..ctor(TargetInfo Target) in c:\Users\pariwat\Documents\Unreal Projects\MyProject2\Plugins\AirSim\Source\AirSim.Build.cs:line 65
--- End of inner exception stack trace ---
at System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
at System.Reflection.RuntimeConstructorInfo.Invoke(BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
at System.RuntimeType.CreateInstanceImpl(BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes, StackCrawlMark& stackMark)
at System.Activator.CreateInstance(Type type, BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes)
at System.Activator.CreateInstance(Type type, Object[] args)
at UnrealBuildTool.RulesAssembly.CreateModuleRules(String ModuleName, TargetInfo Target, FileReference& ModuleFileName)

Timebutt, 2017-02-27:
No, I can't right click on the .uproject yet. Nothing is currently listed for the 'open with' option. This [link ](https://answers.unrealengine.com/questions/464207/generate-visual-studio-project-option-missing.html)(as was mentioned earlier) doesn't help me as I don't seem to find the UnrealVersionSelector-Win64-Shipping executable file. Will keep looking to fix this ...

[image]

Timebutt, 2017-02-27:
No, I can't seem to get that to work. I didn't have any file associations set up in right click menu, so I added one myself. Browsing through the different Unreal binaries in the Unreal x64 Binaries folder (C:\Program Files\Epic Games\UE_4.14\Engine\Binaries), I couldn't find any blue icon so I just associated it with the 'default' UE4Editor.exe. No extra right click options appeared afterwards, so didn't really help me.

[image]

What are you using to open the .uproject file then?

mulysatest, 2017-02-27:
If your icon of Unreal Engine" (in Open with menu) is not show in blue.
It mean once you right click you will not be  see the  generate visual studio project file as mentioned in the tutorials

[image]

It used to have problem like that. All you have to do is choose the right software.
Please add my skype: asylu3 I will help you fix that via teamviewer.

Timebutt, 2017-02-27:
I got my issue fixed and can now generate the Visual Studio project files flawlessly. None of the issues you mention above. Are you running Unreal 14.3? Did the AirSim build finish successfully?

mulysatest, 2017-02-27:
I just noticed that I am using UE 4.15 , now downloading 4.14 now and see if it will be fine or not

mulysatest, 2017-02-27:
@Timebutt, How did you fix the problem?

I did following things but still no luck
- just installed UE 4.14 
- rebuild airsim plugin with buid.cmd and got this result
---------------------------------------------------------------------

              Total    Copied   Skipped  Mismatch    FAILED    Extras
   Dirs :       142         0         1         0         0         0
  Files :       909         8       901         0         0         0
  Bytes :  321.80 m  286.97 m   34.83 m         0         0         0
  Times :   0:00:00   0:00:00                       0:00:00   0:00:00

  Speed :           817692554 Bytes/sec.
  Speed :           46788.743 MegaBytes/min.
  Ended : 28 กุมภาพันธ์ 2017 1:19:19

- create blank c++ project 
- copy plugin from airsim to new UE project
- right click on .uproject and got same error msg.
Discovering modules, targets and source code for project...
ERROR: Unable to instantiate instance of 'AirSim' object type from compiled assembly 'MyProject2ModuleRules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'. Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements. The CLR exception details may provide more information: System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable. See readme: https://github.com/Microsoft/AirSim/blob/master/docs/install_boost.md
at AirSim.AddEigenDependency() in c:\Users\pariwat\Documents\Unreal Projects\MyProject2\Plugins\AirSim\Source\AirSim.Build.cs:line 78
at AirSim..ctor(TargetInfo Target) in c:\Users\pariwat\Documents\Unreal Projects\MyProject2\Plugins\AirSim\Source\AirSim.Build.cs:line 65
--- End of inner exception stack trace ---
at System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
at System.Reflection.RuntimeConstructorInfo.Invoke(BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
at System.RuntimeType.CreateInstanceImpl(BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes, StackCrawlMark& stackMark)
at System.Activator.CreateInstance(Type type, BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes)
at System.Activator.CreateInstance(Type type, Object[] args)
at UnrealBuildTool.RulesAssembly.CreateModuleRules(String ModuleName, TargetInfo Target, FileReference& ModuleFileName)

mulysatest, 2017-02-27:
@Timebutt what is your EINGEN_ROOT value?

mulysatest, 2017-02-27:
I already define EINGEN_ROOT in ENV why I still got this error 
System.Exception: EIGEN_ROOT is not defined

mulysatest, 2017-02-27:
I think I found the problem it was the confusing at this line of code at AirSim.Build.cs: line 78

```
 private void AddEigenDependency()
    {
        string eigenPath = System.Environment.GetEnvironmentVariable("EIGEN_ROOT");
        if (string.IsNullOrEmpty(eigenPath) || !System.IO.Directory.Exists(eigenPath) || !System.IO.Directory.Exists(Path.Combine(eigenPath, "eigen3")))
        {
            throw new System.Exception("EIGEN_ROOT is"+ eigenPath + "/is not defined, or points to a non-existant directory, please set this environment variable.  " +
                "See readme: " + readmurl);
        }
        PrivateIncludePaths.Add(Path.Combine(eigenPath, "eigen3"));
    }

```
then I fixed my EIGEN_ROOT to match their requirement is ok now

lovettchris, 2017-02-27:
Great.

closed 2017-02-27 by lovettchris

Timebutt, 2017-02-28:
@mulysatest: to possibly help future users, what exactly was the issue with your EIGEN_ROOT? You didn't have it defined as ENVIRONMENT_VARIABLE, you had the spelling wrong or ... ?