microsoft/AirSim #65: ArduCopter compatibility?

FRopero, 2017-02-28:
Hi!,
Just a question, there will be an AirSim version compatible with ArduCopter?. I tested it and it does not work.

Thanks

sytelus, 2017-03-01:
Current version is probably not fully compatible with ArduCopter because it uses MavLink differently. However we haven't tested this yet. If you have experience with ArduPilot, please go ahead and try and feel free to contribute changes to make it compatible. All our simulation code is well isolated from any MavLink code as well as Unreal code. So it should be easy to do changes.

madratman, 2017-03-07:
http://ardupilot.org/dev/docs/gsoc-ideas-list.html Let's wait for the summer to end heh

lovettchris, 2017-03-07:
Hey I see they list "Support for AirSim simulator" that's awesome.  Did you do that?  Love to get the Arducopter community help on getting mavlink compatibility with that stack.  The simple stuff probably already works, like takeoff/land, but the offboard control (SET_POSITION_TARGET_LOCAL_NED ) is a bit more complex.  The arducopter implementation of that is probably a subset of px4 at this point...

FRopero, 2017-03-07:
Oh!, It would be awesome to get ArduCopter support in the AirSim simulator.

realdealneil, 2017-11-07:
I see this is closed.  Has there been any development in this area?  I am interested in helping to develop AirSim compatibility for ArduCopter.  Is anyone out there working on this currently?

shubham9436, 2019-01-20:
> One of the Ardupilot devs has started a branch to add ardupilot compatibility here: https://github.com/khancyr/AirSim/tree/ardupilot

has it been developed?

Officepot, 2019-01-22:
I am also very interested in AirSim for Ardupilot. I've looked into the repository by @khancyr but it is not a running product yet. Thanks!

khancyr, 2019-01-22:
hello,
unfortunately, I don't think I will find time to continue this work soon ...

tridge, 2019-05-08:
@sytelus we now have a GSoC student working on AirSim support for ArduPilot. The student is Rajat Singhal, and he is mentored by myself and @peterbarker. 
We just had our kick off meeting where we discussed the approach. There are really three choices:
 - add a MAVLlink based SITL backend to ArduPilot
 - extend the existing ArduCopterSolo backend that is in AirSim (uses UDP and a fixed C structure SensorMessage)
 - add a new UDP interface to AirSim based on extensible structured data, such as JSON or XML

The MAVLink approach has the advantage that we would be able to use existing AirSim versions without modification, but it doesn't look like the MAVLink interface uses the steppable timer, which means the timing is likely to be pretty awful. We try to use what we call "lock-step" scheduling to simulators in ArduPilot, which seems to be equivalent to "steppable" in AirSIm. If we can't do lock-step then we apply a little filter to data to ensure that it is kinematically consistent, plus we replace the state estimator in ArduPilot with one that uses the simulation data directly (avoiding the normal EKF ardupilot uses for position, attitude, velocity etc)

Extending the ArduCopterSolo backend is another approach. That is a bit of an odd one as it uses a data structure (SensorMessage) which we stopped using in ArduPilot a long time ago. We'd need to create a new ArduPilot SITL backend that uses that message structure. We'd also need to add the use of the steppable timer to that AirSim backend.

My favourite option is a new extensible interface, perhaps using JSON or XML, so that we can maintain compatibility across updates to both ArduPilot and AirSim. That will mean a new interface in AirSim which outputs sensor data as a single structured packet encoded with JSON or XML, and that accepts actuator data using the same format (but different fields). Each packet would be timestamped, and we'd use the steppable clock in AirSim.

Comments from the AirSim developers would be very welcome on the right approach, particularly with regard to maintaining good support for ArduPilot long term. Also note that we'd like this to be for more than just multicopters. ArduPilot supports a huge range of vehicle types, and I'm hoping that AirSim will be a good choice for developing new vehicles.

sytelus, 2019-05-13:
Hi @tridge - thank you for taking this initiative. I think the good option would be either to modify [ArduCopterSolo files](https://github.com/microsoft/AirSim/tree/master/AirLib/include/vehicles/multirotor/firmwares/mavlink) or add new ones. The lock-step clock might be easy to support. Currently simple_flight uses SteppableClock and in fact that's the[ default unless we use PX4](https://github.com/Microsoft/AirSim/blob/67f438629c4ca4bcf25436d5e21e9839487a6491/AirLib/include/common/AirSimSettings.hpp#L1076). The way steppable clock works is simple. The clock interfact in AirSim simply provides current time. For steppable clock, one must [advance clock manually](https://github.com/microsoft/AirSim/blob/67f438629c4ca4bcf25436d5e21e9839487a6491/AirLib/include/common/ClockBase.hpp#L46) so who ever asks for current time gets whatever time that was set. So basically, you can [read clock interface](https://github.com/microsoft/AirSim/blob/67f438629c4ca4bcf25436d5e21e9839487a6491/AirLib/include/common/ClockBase.hpp#L17)  to get current time and send it to ardupilot (or diff with last reading and send the diff), Does this makes sense? Please let me know if any questions.

Also, you might be aware about [ArduPilot work](https://github.com/microsoft/AirSim/pull/1387) by @KerryMoffittRtn that is now merged with AirSim.

rajat2004, 2019-07-16:
Hi, I have opened a PR for ArduCopter support - https://github.com/microsoft/AirSim/pull/2075
Has Lock-Step Scheduling with encoding of sensor data in JSON format

It's a new backend, doesn't use the existing Solo API since there were problems with Lock-Step Scheduling with it.
Please have a look at it and review!

auturgy, 2020-04-12:
Suggest closed by https://github.com/microsoft/AirSim/pull/2075

rajat2004, 2020-04-12:
Yup, thanks for reminding! Support for Copter, Rover is there in ArduPilot & AirSim master, as well as releases 1.3.0 and later
Details on setup and usage are on this wiki page - https://ardupilot.org/dev/docs/sitl-with-airsim.html

closed 2020-08-10 by jonyMarino