microsoft/AirSim #66: Problem connecting to Pixhawk (SITL PX4 simulation)

Timebutt, 2017-02-28:
Hi,

I'm currently trying to get my setup running without having a Pixhawk connected, so I'm following the [SITL instructions](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md#software-in-loop-simulation-sitl). So far, I've set up the entire Linux environment required, built the posix SITL PX4 version and started it on my machine. This is what it looks like, running on my machine:

[image]

As per the instructions, I have edited the settings.json file (located in Documents/AirSim, where I cloned the AirSim project). My file is currently showing this (shown is the entire settings.json content):

[image]

Comparing this to the settings.json used in [this instructional video](https://youtu.be/HNWdYrtw3f0?t=380), I notice quite a difference here. Are these settings required for correct functionality of AirSim, and why would these be missing with me?

[image]

In any case, I have set up an Unreal project as described in [this video](https://www.youtube.com/watch?v=1oY8Qu5maQQ&feature=youtu.be). Now, when I launch the Unreal in SimGameMode, my machine renders about one second of video and then Unreal completely freezes. In the top left corner, it's saying to check my json settings, and that it can't connect to my PX4 simulation for some reason. When I look at the output of my bash, no messages are appearing, so no connection is being set up.

[image]

So, two questions:

1. Is there any way to get a more detailed output of exactly what's going wrong? Am I missing something here? Why can't my Unreal environment connect to the Pixhawk simulation?

2. Why does my Unreal editor freeze after about 1s of rendering? I only get this when using the SimGameMode, the other modes are just fine and it can't be my machine (Intel i7 6900K, Nvidia GTX980 TI, 16Gb RAM, SSD, ...). Is this because no connection can be set up to the PX4?

All help is greatly appreciated!

lovettchris, 2017-03-01:
This is probably our bug, we did some big refactoring recently, I'll look into this right away, thanks for reporting it.

Timebutt, 2017-03-01:
Ok great news!
Keep me posted when the issue gets fixed, I will verify straight after.

sytelus, 2017-03-01:
@Timebutt I'm testing the fix and would push it soon.

fmzhu, 2017-03-01:
so can anyone make a video to show how to realize the SITL with gamepad ? I'm stuck at it too! Thank you very much

Timebutt, 2017-03-01:
The setup I'm trying to run is using SITL and a PS3 controller (may have to emulate an Xbox 360 controller, don't know yet). Once this issue is fixed and I get everything running, I will be contributing to the documentation on getting it up and running.

sytelus, 2017-03-02:
@Timebutt @lovettchris I've pushed the change to put back missing settings in json file. So you should be now be able to modify LocalHostIp as in the video.

Timebutt, 2017-03-02:
Hi @sytelus, thanks for committing the fixes.
Unfortunately, I can't verify the functionality yet because I'm still experiencing the crashing Unreal Editor environment. The good news however, is that I found out what's exactly going on. The environment isn't crashing, but rather rendering **extremely** slow. I started looking through the logs and found this:

[image]

So apparently Unreal Editor finds three eligible GPUs, and decides somehow to always select the Intel HD530 GPU. This is why I can perfectly use the other game modes (but even there I get sub par frame rate, the HD530 isn't exactly a powerful GPU), but as soon as I start the SimGameMode the HD530 can't keep up and generates one frame per minute (or even less!). 

I found [this possible solution](https://forums.unrealengine.com/showthread.php?3905-Multiple-Graphics-Adapters-(related-to-4-1-QFE-very-low-fps-rendering)), written by an Unreal Engine developer but can't seem to get it right just yet. 

I'm glad I found out why my environment is crashing, now to find the solution. As soon as I get everything working, I will write down extensively what I have found, hopefully to help other people. And then to verify the fixes for the SITL communication problem ;)

lovettchris, 2017-03-02:
You will also need [this fix](https://github.com/Microsoft/AirSim/commit/96b2cdd8c51b2994b25c269cfa85972cbe74f3c5).

Timebutt, 2017-03-02:
Confirmed: the two fixes correct the SITL functionality!
The drone correctly connects to virtualized PX4, as can be seen in the following screenshot:

[image]

Thank you very much for the quick fixes @lovettchris and @sytelus! This will definitely help other people out as well. Issue can be closed.

Now to find out why the Unreal Editor considers my GTX 980 TI an expensive paperweight and prefers to run the application on the integrated GPU ...

closed 2017-03-02 by Timebutt