microsoft/AirSim #68: Undefined symbol at runtime when starting simulation in Unreal (Linux)

micmancini16, 2017-03-01:
Hi,

I've been working on installing AirSim on a Linux system. I managed to compile and install the plugin on an empty Unreal project, trying to set up a SITL simulation.
Problems arise when I switch to Game Mode = SimGameMode and press Play. Unreal instantly crashes, returning on terminal the following error:

_symbol lookup error: /media/isarlab/e1a54258-46b5-46ed-81f7-5cf01e504c48/UnrealProjects/AirSimProject/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so: undefined symbol: _ZN3msr6airlib8Settings12loadJSonFileENSt3__112basic_stringIwNS2_11char_traitsIwEENS2_9allocatorIwEEEE_

I compiled AirSim through this script [https://github.com/Microsoft/AirSim/blob/master/build.sh](url), although I had to comment lines 83-88, as it was not able to find lib*.a files in the given directories, and manually locate and move compiled libraries on the required target directories. Furthermore, my Unreal project starts regularly when selecting default GameModes.

Has anybody had success on using AirSim on Linux yet? Any help is appreciated!

bkueng, 2017-03-02:
@micmancini16 https://github.com/Microsoft/AirSim/pull/72 should fix this.

micmancini16, 2017-03-02:
Thanks @bkueng, I'll check it out as soon as possible!

micmancini16, 2017-03-02:
@bkueng I tried to use the flag -stdlib=libc++ but I got the following compiler errors:
`/usr/include/c++/v1/string:1938:44: error: 'basic_string<_CharT, _Traits, _Allocator>' is missing exception specification 'noexcept(is_nothrow_copy_constructible<allocator_type>::value)'
basic_string<_CharT, _Traits, _Allocator>::basic_string(const allocator_type& __a)`.
To fix those I had to change the flag in -stdlib=libstdc++ both in external/rpclib make file and in the line
`set(CMAKE_CXX_FLAGS "-stdlib=libstdc++ ${CMAKE_CXX_FLAGS}")`
of the file CommonSetup.cmake in cmake-modules.
This way I was able to compile everything but again, when I start the simulation in Unreal everything crashes and gives me this error (different from the previous one, but still related to AirSim)
`symbol lookup error: /media/isarlab/e1a54258-46b5-46ed-81f7-5cf01e504c48/UnrealProjects/TestAirSim-4.14/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so: undefined symbol: _ZN12common_utils10FileSystem22getUserDocumentsFolderEv`

micmancini16, 2017-03-02:
UPDATE: I managed to compile everything with the flag -stdlib=libc++ by adding the flags:
`-DCMAKE_C_COMPILER=gcc-6 -DCMAKE_CXX_COMPILER=g++-6`
in the build.sh script. However, I still get the same error when trying to start the simulation within Unreal: 
`symbol lookup error: /media/isarlab/e1a54258-46b5-46ed-81f7-5cf01e504c48/UnrealProjects/TestAirSim-4.14/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so: undefined symbol: _ZN12common_utils10FileSystem22getUserDocumentsFolderEv`

bkueng, 2017-03-02:
>  I tried to use the flag -stdlib=libc++ but I got the following compiler errors:
/usr/include/c++/v1/string:1938:44: error: 'basic_string<_CharT, _Traits, _Allocator>' is missing exception specification 

This seems to be a bug in Ubuntu: http://stackoverflow.com/questions/37096062/get-a-basic-c-program-to-compile-using-clang-on-ubuntu-16

You cannot use libstdc++ (or GCC which uses libstdc++, I don't know if it can use libc++), because you get the symbol errors as you describe (read https://github.com/Microsoft/AirSim/pull/43 why). The solution is to use clang and libc++ (maybe there are others, like building the Unreal Engine with GCC, but I haven't tried that).

I'm on Fedora and have it working with clang and libc++. I had lots of symbol problems as well until I used exactly this combination.

micmancini16, 2017-03-02:
Ok, we managed to make everything work.
By reading the page suggested by @bkueng we found that libc++ version has a bug (https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=808086) which has been fixed in version 3.9.
Thus, we had to remove the libc++ 3.7 version and install the 3.9 version (which is not available in the official repo yet) that can be downloaded from 
https://launchpad.net/ubuntu/+source/libc++
After rebuilding everything, we finally could run the plugin with no errors.

Thanks a lot @bkueng

closed 2017-03-02 by micmancini16