microsoft/AirSim #69: Loading AirSim plugin into Unreal Project in Linux

saihv, 2017-03-01:
I have managed to build Unreal Editor and Airsim for Linux, and I have the Unreal/Plugins folder in my AirSim directory. I tried to create an empty C++ project, copy the Plugins directory and change the .uproject file as described in the instructions: but I am not sure what to do next. I tried running Unreal Editor directly, but it complains 

`LogInit:Warning: Incompatible or missing module: libUE4Editor-AirSimProject.so
LogInit:Warning: Incompatible or missing module: libUE4Editor-AirSim.so
`
It then tries to build them and fails. I tried running ./GenerateProjectFiles.sh with the project directory as an argument, but then it fails with the message 

`ERROR: Unable to instantiate instance of 'AirSim' object type from compiled assembly 'AirSimProjectModuleRules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.  Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements.  The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable.  See readme: https://github.com/Microsoft/AirSim/blob/master/docs/install_boost.md
`
This is strange because both EIGEN_ROOT and BOOST_ROOT are set, and the AirSim build wouldn't have succeeded without those anyway. Has anyone succeeded in importing the Airsim plugin in Linux?

micmancini16, 2017-03-01:
I managed to solve that problem by manually editing the file AirSim.build.cs in the AirSim/Source directory (inside the Plugins directory of your Unreal project). In that file, I changed the line 
`string eigenPath` inside the function `AddEigenDependency()` hardcoding the path to my eigen directory. Similarly, I did it for boost by editing the path in `AddBoostDependency`. This is just a workaround but in this way you should be able to compile the project.

inkjet, 2017-03-02:
An alternative is to check installation directory with `echo $EIGEN_ROOT`.  Mine was pointing to `/home/user/eigen3`, when it should have actually been `/home/user/eigen3/eigen3`

Still, I'm having the same problems as @saihv , where following the given compilation instructions only creates .a files in `AirSim/Unreal/Plugins/AirSim/Source/AirLib/lib/x64/Debug`. What steps did you follow to create the `libUE4Editor-AirSim.so` file, @micmancini16 ?

chenste, 2017-03-02:
I was able to create the `libUE4Editor-AirSim.so` by manually copying over the .a files to the correct path. However even after creating the file, Unreal Engine crashes when using SimGameMode as detailed in #68

saihv, 2017-03-02:
Thanks, @micmancini16! That worked, and I was able to compile the project. 

Then I ran UE4Editor with the uproject file as its argument. First it complained that the .so files are missing and whether they should be built, and I said yes: after which it didn't throw any more errors but I can't find where the .so files are located. But then it seems like it's building the Editor again: it started compiling some 908 files from the Editor folder. Guess I'll have to wait for this to figure out if it worked or not.

micmancini16, 2017-03-02:
Hi @inkjet, the file 'libUE4Editor-AirSim.so' is created only when you compile a new Unreal project with the Airsim plugin in it. 
@saihv I experienced the same situation: as I compile the project for the first time, it recompiles more than 900 Editor files. Later, it will recompile only the Airsim-related files. Let us know what is your outcome when you're done!

saihv, 2017-03-02:
Had a couple of issues with librpc.a etc. not being in the right locations, so still haven't built it completely. (I had to manually run make in the rpc folder to get those libraries). Every time the AirSim project fails to compile, it breaks the editor too: so I'm having to rebuild the editor and then Airsim. Will keep you posted!

inkjet, 2017-03-03:
@saihv - I hit the same upper/lower case issue as you just mentioned. It's been slow going since then, rebuilding the editor and AirSim.  Have you been successful?

Also, I had the same issue with the AirSim plugin *.a files not being in the right place -- example `clang: error: no such file or directory: '/home/user/Documents/Unreal Projects/ExampleCpp/Plugins/AirSim/Source/AirLib/deps/MavLinkCom/lib/libMavLinkCom.a'` The files can be found in `AirSim/cmake/output/lib`. Did everyone else have to manually copy them over?

saihv, 2017-03-03:
I had that problem when I followed the linux_build.md instructions, but then I abandoned that instructions file and started following what's in the build.sh file. There's a list of cp commands in the sh file that need to be executed that will move the .a files to the right places (but one of them has the wrong syntax, so instead of copying a library into AirLib/deps/MavlinkCom/lib/, it moves it to MavlinkCom and renames the library itself as lib).

saihv, 2017-03-03:
Update: I was finally able to successfully build the plugin within my Unreal project. I opened the project, selected SimGameMode and then clicked on Play, but then the Editor crashed with the message

saihv, 2017-03-03:
I have the same "undefined symbol" error after building it with libc++3.9, which I installed from https://launchpad.net/ubuntu/+source/libc++/3.9.1-2. @micmancini16, could you please confirm if this is the version you used for your build?

micmancini16, 2017-03-03:
@saihv Yes, those are the same libc++ and clang versions that I used. I expicitely removed the libc++ 3.7 version before installing the updated one, to be sure that clang used the correct libraries. That was enough to solve the problem for me.

saihv, 2017-03-03:
Built it again from scratch with clang and libc++3.9, and it works now. Thanks for the help!

closed 2017-03-03 by saihv

inkjet, 2017-03-04:
@saihv Glad to hear you got it working!  I'm encountering the same issues you faced, and this seems to be the final one I need to get over the hump.  

How did you find the version of libc++ you had?  I can't find the install location, and want to upgrade to 3.9.

saihv, 2017-03-04:
@inkjet

The version of libc++ you need is 3.9. You can check what version you have by running dpkg -s libc++-dev. If it's lower than 3.9 (which I am guessing it is) all you should have to do is remove existing versions (remove others too, such as libc++abi) and visit this page: https://launchpad.net/ubuntu/+source/libc++/3.9.1-2.

From this page, you'll have to download deb files for libc++-dev, libc++1, libc++abi1, libc++abi-dev and libc++-helpers (the reason there are so many is that some of them are dependencies for others). Hope this helps.

inkjet, 2017-03-07:
@saihv I'm still trying to get this to work on Ubuntu 16.04. I've updated all the relevant libc++ versions to 3.9 - for example, `dpkg -s libc++abi1` returns

`Package: libc++abi1 
Status: install ok installed
...
Source: libc++
Version: 3.9.1-2`

And this is the same for all the packages.  But the built AirSim library still crashes when I hit play due  the symbol errors.  When I look at the built .so file with `ldd -r -v /home/inkjet/Documents/Unreal\ Projects/Quad/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so`, I see all the `undefined symbol: _ZN12common_utils10FileSystem22getUserDocumentsFolder` errors.

I seem to be stuck at this point.  Any idea what I might be missing that got you over the hump?

lovettchris, 2017-03-07:
FileSystem::getUserDocumentsFolder is defined in ~\AirLib\src\controllers\FileSystem.cpp, can you check it is included in your cmake/AirLib/CMakeLists.txt ?

saihv, 2017-03-07:
@inkjet

I had the same symbol error pop up even with the right libc++ versions, but then I realized I forgot to make sure to build the plugin with Clang. Just to be sure: are you using build.sh, or did you use something like 

`
export CC=/usr/bin/clang
export CXX=/usr/bin/clang++`

inkjet, 2017-03-08:
@saihv  

I'm using the build.sh found in the `/cmake` folder, then copying over the built *.a files to the proper directories.  Should I set the Clang variables first?

saihv, 2017-03-08:
I think so, because the build.sh in /cmake doesn't do that. What I would suggest is to follow the steps in the build.sh located in the root directory (not the one in the cmake folder). You'll notice that it uses clang for compiling the plugin as opposed to g++/normal make.

inkjet, 2017-03-08:
@saihv 

That did it! I edited the `build.sh` file in the `cmake` folder to export the `CC` and `CXX` variables as you mentioned above. (For those who are ignorant like me, this [tells make to use clang](http://stackoverflow.com/questions/7031126/switching-between-gcc-and-clang-llvm-using-cmake) instead of the default gcc).

Now my drone sim is up and flying in Linux.  Thanks.

lovettchris, 2017-03-08:
awesome, can you push back any cmake/build.sh changes you made as a pull request?  I'd like to merge them back and rationalize the two build scripts we have into one...

saihv, 2017-03-08:
@inkjet That's great! Just curious, does it work smoothly when you press Play in SimGameMode? I (and a few others, based on comments here) had to comment out a mutex lock guard statement in the plugin's code which was causing the simulation to freeze up.

inkjet, 2017-03-09:
@saihv The gameplay is fairly choppy in when I hit play, and slows down to an almost-unusable 1 fps when in record mode.  However, I did experience much better performance running in DebugGame Editor mode (to those who aren't familiar, add "-game -debug" to the command line when launching).  Any hints on optimizing performance?

@lovettchris I will roll up my changes in a pull request shortly.