microsoft/AirSim #71: Windows 10 with driver signature enforcement

Zhao1ming, 2017-03-01:
The driver of PX4 can not work in Win 10

lovettchris, 2017-03-01:
What error are you seeing?  We are using http://qgroundcontrol.com/ which bundles a px4 driver that works on windows 10.  Can you try qgroundcontrol and see if it will connect to your pixhawk ?

Zhao1ming, 2017-03-01:
Yeah, I use it ,but when I plug in the pixhawk by USB port, there is nothing happened. It looks like the driver can not work. QGC can not automatically respond to the pixhawk.

Zhao1ming, 2017-03-01:
[image]

Zhao1ming, 2017-03-01:
[image]

lovettchris, 2017-03-02:
Hmmm, ok, do you hear USB sounds from Windows saying it recognizes your device? And do you see a new device show up in Device Manager under Ports (COM & LPT), like this?  If you don't hear a sound, try a different USB cable (you might have a power only cable instead of a data cable?).

[image]

Zhao1ming, 2017-03-02:
[image]

Zhao1ming, 2017-03-02:
I can hear the voice of the machine, but there is nothing about the system. Looks like no driver

lovettchris, 2017-03-02:
Interesting.  I have not tried that particular hardware, perhaps it needs to have the firmware flashed.  See the video on this page for initial configuration including [updating the firmware](https://dev.px4.io/starting-initial-config.html).   Now you also will need to install the [PX4 version 1.4.4](https://github.com/PX4/Firmware/releases/tag/v1.4.4).

closed 2017-04-05 by lovettchris