microsoft/AirSim #73: Using Joystick/Gamepad (Alternative to RC)

fmzhu, 2017-03-02:
hi, I'm trying to run is using SITL and a Joystick controller. Wenn I connect it to the QGroundControl, nothing happens, shown as below
[image]

the first three steps are well done, I'm stuck at step 4, so my question is:  do we need special controller, or any Joystick can work? thank you very much!

1. Connect Joystick/Gamepad such as XBox 360 controller to PC. Start QGroundControl.
2. Click on the purple "Q" icon on left in tool bar to reveal the Preferences panel.
3. Go to General tab and check the Virtual Joystick checkbox.
4. Go back to settings screen (gears icon), click on Parameters tab, type COM_RC_IN_MODE in search box and change its value to either Joystick/No RC Checks or Virtual RC by Joystick.
5. You should now see a new tab "Joystick" where you can do the calibration.

Timebutt, 2017-03-03:
Hi @fmzhu, I think we'll be needing a bit more information on how much of the entire environment you have running right now.

Do you have the PX4 instance running on your machine, and an active Unreal Editor project? Does your Unreal environment correctly connect to the PX4 simulator? You should see a lot of lines appearing, just like this:

[image]

I believe having a correct connection from the Unreal simulation to the PX4 is a requirement to be able to set everything up in QGroundControl (devs, correct me where I'm wrong!). There are images available that explain the AirSim architecture and how everything is connected, I'll see if I can dig those up to illustrate what I'm trying to explain.

alexferra, 2017-03-08:
I'm having the same issue. I'm trying to use an Xbox 360 controller and I don't see this option:

4. Go back to settings screen (gears icon), click on Parameters tab, type COM_RC_IN_MODE in search box and change its value to either Joystick/No RC Checks or Virtual RC by Joystick.

Timebutt, 2017-03-09:
Hi @alexferra, you won't see the joystick option before you have the entire AirSim environment up and running. You need the PX4 SITL running in bash and an active simulation in Unreal Editor 4.14.3. Only then will the _joystick_ option (and a lot more really) appear in QGroundControl. 

To verify you have the environment correctly set up, you need to check your PX4 bash for output like in the screenshot I posted above. That way, you know the simulator is running and correctly connected to PX4.

closed 2017-04-19 by sytelus

gregoryps1989, 2017-06-12:
I also stuck at the same part. I have the AirSim environment, QGroundControl and PX4. Can you explain how to bring these together? When I start PX4 it stops at 'waiting for initial data on UDP port 14560. Please start the flight simulator to proceed.' 
In QGroundControl is no Joystick recognized.

Timebutt, 2017-06-12:
Hi @gregoryps1989, sounds like you need to start AirSim in _AirSimGameMode_ if you see the '_waiting for initial data on UDP port 14560_' message on PX4. QGroundControl won't show you any joystick option if there is no connection to a vehicle (which is the case in your situation), so you need to have the AirSim simulation running first.

My suggestion, watch the [instructional videos](https://github.com/Microsoft/AirSim#video-tutorials) Chris has made. I have also written [an article](https://timebutt.github.io/static/using-a-phantom-dji-controller-in-airsim/) explaining how to use a Phantom 2 controller with the AirSim simulation, [about halfway down](https://timebutt.github.io/static/using-a-phantom-dji-controller-in-airsim/#tryingitoutinairsim) I specify the order in which the different steps should be executed to get everything running correctly.

Let us know if this helps!

gregoryps1989, 2017-06-12:
Hi @Timebutt, thank you for your quick response. I guess I have trouble making a correct 'settings.json'-file. At first i used the standard code from the instruction [site](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md), but starting AirSim it complains not finding a correct flight controller. 
Then I copied your code from [here](https://github.com/Microsoft/AirSim/issues/139) and now it works somehow.

lovettchris, 2017-06-12:
Right the key there was "UseSerial": true is for direct serial port connection to Pixhawk hardware, whereas you are using the SITL mode so you needed "UseSerial": false.