microsoft/AirSim #76: error C2220

BlackFH, 2017-03-03:
Hello,I follow the steps but encountered an error c2220 while running build.cmd.
and I try to disable treat warnings as errors but it`s not work.
Is that because my system language is Chinese?
[image]

[image]

clovett, 2017-03-03:
Looks like the file contains a character that cannot be returned in your code page, see https://msdn.microsoft.com/en-us/library/ms173715.aspx.  Can you try saving that for (half.h) as utf8 ?

lovettchris, 2017-03-03:
I have verified that this file from the Eigen library does indeed contain characters that should have been encoded at UTF-8.  This is how I did it.  When I use Visual Studio to save the file with ansii encoding like this:
[image]
then you will see Visual Studio correctly report the error:
[image]

But if you save it with this encoding then everything is fine, and this should work on your machine.
[image]

But the characters in the file that are causing this problem are probably in every single Eigen file because it is part of the copyright header.  Specifically it is the special curly quotes used in this part of the header:

> “AS IS”

If you replace the curly quotation characters with simple double quotes "" then that should also fix it.

Technically this is a bug in the Eigen library, they should have UTF-8 encoded all their files.  So if you have a tool that can utf-8 encode all the Eigen files then I would use that just in case there are any other special characters used anywhere.  Trick is that tool probably needs to load them as Western European Codepage 28591, and not the default code page of the system.  It is your default code page that does not contain those special characters.

BlackFH, 2017-03-04:
Thanks Chris,also thanks for Clovett's reply.It's strange that Visual Studio will still report the error after I saved the file with UTF-8 encoding.However,the problem is solved after I replaced the curly quotation characters.I'm sorry about my poor English.I will close this issue.

closed 2017-03-04 by BlackFH

clovett, 2017-03-04:
You're English is fine, thanks for closing.   I'll follow up with vc complete this on why this happened.  Turns out the file was utf8 expires it just doesn't have a bite order Mark....

lovettchris, 2017-03-06:
By the way there is a way to fix this using /utf-8 compiler flag, if you add this to the mSVC section of the CMakeLists.txt: 
       set(RPCLIB_BUILD_FLAGS "/utf-8")

More info here: [https://forum.kde.org/viewtopic.php?f=74&t=139230](https://forum.kde.org/viewtopic.php?f=74&t=139230).

whxru, 2018-08-31:
It's not ok for me to save half.h with encoding UTF-8, but UTF-8 with BOM works correctly.