microsoft/AirSim #78: Arming Denied：Preflight check have failed.

zf223669, 2017-03-04:
Hi, I have met a problem that i used the DroneShell to control the drone, however, when i inputted the command arm, it showed that Arming Denied：Preflight check have failed.

sytelus, 2017-03-09:
Its most likely a set up problem. I have seen this happening before but can't recall the context. The way to debug this issue is simple: Use QGroundControl to connect to Pixhawk and see if there is anything red. Then try to arm and takeoff from within QGroundControl. If that works then it should work in AIrSim as well. Also, try out latest code (there were few API issues were resolved recently).

closed 2017-03-09 by sytelus

TomorrowIsAnOtherDay, 2017-06-08:
Then try to arm and takeoff from within QGroundControl.

TomorrowIsAnOtherDay, 2017-06-08:
How to arm and takeoff from within QGroundControl?

lovettchris, 2017-06-08:
this thread is actually closed already... please open a new one, but QGroundControl probably has it's own forums somewhere.