microsoft/AirSim #79: Linux:

patrickelectric, 2017-03-04:
OS: `Linux HP 4.9.11-1-ARCH`
clang: `clang version 3.9.1 `
unreal: `4.15.0-release`

- First Message
[image]
- Second Message
[image]

```
~/git/UnrealEngine/Engine ~/git/UnrealEngine
Using clang version '3.9.1' (string), 3 (major), 9 (minor), 1 (patch)
Creating makefile for MyProject (no existing makefile)
ERROR: Couldn't find target rules file for target 'MyProject' in rules assembly 'UE4Rules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.
Location: /home/patrick/git/UnrealEngine/Engine/Intermediate/Build/BuildRules/UE4Rules.dll
Target rules found:
	UE4Client - /home/patrick/git/UnrealEngine/Engine/Source/UE4Client.Target.cs
	UE4Editor - /home/patrick/git/UnrealEngine/Engine/Source/UE4Editor.Target.cs
	UE4Game - /home/patrick/git/UnrealEngine/Engine/Source/UE4Game.Target.cs
	UE4Server - /home/patrick/git/UnrealEngine/Engine/Source/UE4Server.Target.cs
	BlankProgram - /home/patrick/git/UnrealEngine/Engine/Source/Programs/BlankProgram/BlankProgram.Target.cs
	BuildPatchTool - /home/patrick/git/UnrealEngine/Engine/Source/Programs/BuildPatchTool/BuildPatchTool.Target.cs
	CrashReportClient - /home/patrick/git/UnrealEngine/Engine/Source/Programs/CrashReporter/CrashReportClient/CrashReportClient.Target.cs
	MinidumpDiagnostics - /home/patrick/git/UnrealEngine/Engine/Source/Programs/CrashReporter/MinidumpDiagnostics/MinidumpDiagnostics.Target.cs
	CrossCompilerTool - /home/patrick/git/UnrealEngine/Engine/Source/Programs/CrossCompilerTool/CrossCompilerTool.Target.cs
	UnrealLaunchDaemon - /home/patrick/git/UnrealEngine/Engine/Source/Programs/IOS/UnrealLaunchDaemon/UnrealLaunchDaemon.Target.cs
	DsymExporter - /home/patrick/git/UnrealEngine/Engine/Source/Programs/Mac/DsymExporter/DsymExporter.Target.cs
	ShaderCacheTool - /home/patrick/git/UnrealEngine/Engine/Source/Programs/Mac/ShaderCacheTool/ShaderCacheTool.Target.cs
	UE4EditorServices - /home/patrick/git/UnrealEngine/Engine/Source/Programs/Mac/UE4EditorServices/UE4EditorServices.Target.cs
	UnrealAtoS - /home/patrick/git/UnrealEngine/Engine/Source/Programs/Mac/UnrealAtoS/UnrealAtoS.Target.cs
	ShaderCompileWorker - /home/patrick/git/UnrealEngine/Engine/Source/Programs/ShaderCompileWorker/ShaderCompileWorker.Target.cs
	SlateViewer - /home/patrick/git/UnrealEngine/Engine/Source/Programs/SlateViewer/SlateViewer.Target.cs
	SymbolDebugger - /home/patrick/git/UnrealEngine/Engine/Source/Programs/SymbolDebugger/SymbolDebugger.Target.cs
	TestPAL - /home/patrick/git/UnrealEngine/Engine/Source/Programs/TestPAL/TestPAL.Target.cs
	UnrealCEFSubProcess - /home/patrick/git/UnrealEngine/Engine/Source/Programs/UnrealCEFSubProcess/UnrealCEFSubProcess.Target.cs
	UnrealCodeAnalyzer - /home/patrick/git/UnrealEngine/Engine/Source/Programs/UnrealCodeAnalyzer/UnrealCodeAnalyzer.Target.cs
	UnrealFileServer - /home/patrick/git/UnrealEngine/Engine/Source/Programs/UnrealFileServer/UnrealFileServer.Target.cs
	UnrealFrontend - /home/patrick/git/UnrealEngine/Engine/Source/Programs/UnrealFrontend/UnrealFrontend.Target.cs
	UnrealHeaderTool - /home/patrick/git/UnrealEngine/Engine/Source/Programs/UnrealHeaderTool/UnrealHeaderTool.Target.cs
	UnrealLightmass - /home/patrick/git/UnrealEngine/Engine/Source/Programs/UnrealLightmass/UnrealLightmass.Target.cs
	UnrealPak - /home/patrick/git/UnrealEngine/Engine/Source/Programs/UnrealPak/UnrealPak.Target.cs
	UnrealVersionSelector - /home/patrick/git/UnrealEngine/Engine/Source/Programs/UnrealVersionSelector/UnrealVersionSelector.Target.cs
	UnrealWatchdog - /home/patrick/git/UnrealEngine/Engine/Source/Programs/UnrealWatchdog/UnrealWatchdog.Target.cs
	BootstrapPackagedGame - /home/patrick/git/UnrealEngine/Engine/Source/Programs/Windows/BootstrapPackagedGame/BootstrapPackagedGame.Target.cs

LogInit:Warning: Still incompatible or missing module: libUE4Editor-MyProject.so
LogInit:Warning: Still incompatible or missing module: libUE4Editor-AirSim.so
```

saihv, 2017-03-05:
I think the only supported Unreal version is 4.14. Perhaps you should clone the 4.14 branch of Unreal Editor and then try to build the plugin.

patrickelectric, 2017-03-07:
Someone can explain why it's only working with 4.14 ?

lovettchris, 2017-03-07:
Only because that's what we started with and it is real work to switch to 4.15, API changes in Unreal, etc...  If someone wants to do that work, it would be great :-) hint hint...

patrickelectric, 2017-03-07:
OK, Ill take a look.

lkumar93, 2017-03-10:
I keep getting the same error listed here. It keeps happening for every c+= project I create/open in Unreal in Linux

sytelus, 2017-03-16:
Hi folks - I've updated AirSim to use 4.15. Unfortunately the uasset files after update are no longer compatible with 4.14. I've filed issue at Unreal forums but unless they come up with solution, the minimum required engine version is now 4.15. I've put some code that detects it and shows warning message. I've also added a [quick doc](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_upgrade.md) on how to upgrade your Unreal from from 4.14 to 4.15.

closed 2017-04-19 by sytelus