microsoft/AirSim #82: Binarie Distribution Chain : RAspberryPI p2p Network.

MrNavette, 2017-03-06:
With only a few we could distribute a lot.

Just select who's gonna have the binaries (make private repo: linux host)

Secured servers with secured private p2p links, that could allow you to control how your binaries are spread.

It could help a lot, thoose who got one of these tiny pc could build up this in an eyebat.

**In case you ask thi comes frome there : [README.md ? ](https://github.com/Microsoft/AirSim/blob/master/docs/use_precompiled.md)**

lovettchris, 2017-03-06:
Sorry this is still too cryptic.  Are you suggesting we distribute Unreal Environments using Raspberry PI p2p network ?  I don't see how that would work,  Can you provide more info, what is this Raspberry PI p2p network ?

MrNavette, 2017-03-06:
I mean some of us could share http links or ftp or whatever privates link, hosted on 30$ pc and a 10$ SD card should be fine, for a short period of time. 

So you guys can build real end dowload platform :) And we could have some windows based work platform  in little time.

MrNavette, 2017-03-06:
Gamers don't like linux :) Neithier does the 3D artists.

sytelus, 2017-03-09:
We have mostly zeroed in our binary distribution strategy however we want to wait until we have next version up and running. Hopefully, not too long :).

PS: Let's keep titled and description informative!

closed 2017-03-09 by sytelus