microsoft/AirSim #83: Mutex contention causing freezes in ScheduledExecutor

staylor774, 2017-03-06:
Hi AirSim guys,

I'm seeing some horrendous freezes (10-20+ seconds at a time) that appear to be caused by contention on a mutex. Using VS' Performance Profiler, I've tracked it down to this lock call:

```
void ASimModeWorldBase::Tick(float DeltaSeconds)
{
    world_.lock();  // <-- Blocks for 10-20 sec
```

The only other contention on that mutex is performed in ScheduledExecutor::executorLoop, ln 126:

```
    //TODO: add ability to disable use of mutex
    std::lock_guard<std::mutex> locker(mutex_);
    ...
    sleep_for(sleep_dt);
```

If I comment out this lock_guard, it fixes the freezes, and I haven't noticed any bad side effects yet (from my limited testing). What is the reason for sleeping while the mutex is acquired?

I've reproduced the issue on two machines: Windows 7 and Windows 10, both running VS 2015 Enterprise (14.0.25431.01 Update 3). I don't have any hardware or controller connected. I also noticed that I could reduce the duration of the contention by changing the active configuration from "Development Editor" to "DebugGame Editor"... however I'm not sure why that would make a difference.

Thanks for your work and your support!

Timebutt, 2017-03-06:
Running in _Development Editor_ I sometimes get the exact same scenario you are describing. I can confirm that switching to the _DebugGame Editor_ fixes these horrible freezes for me. In fact, I'm only experiencing the freezes very rarely in this mode.

saihv, 2017-03-06:
I don't know if it's exactly because of this piece of the code, but I can also confirm that Development Editor gives me horrible performance, and DebugGame Editor is much smoother. I am going to test commenting that statement out, thanks for pointing it out.

sytelus, 2017-03-09:
Thanks for reporting this issue. I've almost exclusively use DebugGame Editor during development so haven't see this issue but I'll test out the Development Editor as well. 

Here are some insignts...

So there are two main threads: one that runs high frequency physics engine @ 500Hz and other thread that does rendering at ~30-60 Hz. When it comes time to do rendering, we have to get data from physics engine and use that in rendering. To avoid getting malformed data, we have to use some sort of thread synchronization. Currently this synchronization is simplistic which is basically to stop physics engine thread while rendering thread gets pieces of data out of it. I noticed looking the code that we can optimize it much better so this contention is less. However in longer term, we can probably use fine grained synchronization instead of current course grain and also take advantage of the fact that render thread is only reading the data. 

If you remove the lock, generally things should be still fine as render thread only does reading however once in a while it might read partial update from one step of physics mixed with next step. For example, render thread might get vector component x, y and then get cut off with z being stall. This shouldn't be end of the world because these values typically change slowly anyway. So things should mostly work but I wouldn't bet on it.

sytelus, 2017-08-14:
This should be fixed now. The lock in ScheduledExecutor::executorLoop is only maintained during callback (previously it was through the sleep).

closed 2017-08-14 by sytelus