microsoft/AirSim #84: crash probelm on windows

g-ch, 2017-03-06:
I tried to establish the airsim environment on win10. I followed the instructions. compile went through successfully. simulation mode was selected. but when i pressed play button on virtual engine, the program crashed. here is the report. hope someone could help me. thanks. 
(i have tried on engine vision 4.14 and 4.15. the error reports are the same.)

[image]

C++ EH exception - code e06d7363 (first/second chance not available)

KERNELBASE
VCRUNTIME140
UE4Editor_AirSim!common_utils::FileSystem::createTextFile() [c:\multicopter\airsim\airsim\airlib\include\common\common_utils\filesystem.hpp:113]
UE4Editor_AirSim!msr::airlib::Settings::saveJSonFile() [c:\users\user\documents\unreal projects\first\plugins\airsim\source\airlib\include\controllers\settings.hpp:61]
UE4Editor_AirSim!msr::airlib::Px4QuadX::getConnectionInfo() [c:\users\user\documents\unreal projects\first\plugins\airsim\source\airlib\include\vehicles\configs\px4quadx.hpp:133]
UE4Editor_AirSim!msr::airlib::Px4QuadX::createMavController() [c:\users\user\documents\unreal projects\first\plugins\airsim\source\airlib\include\vehicles\configs\px4quadx.hpp:60]
UE4Editor_AirSim!msr::airlib::Px4QuadX::setup() [c:\users\user\documents\unreal projects\first\plugins\airsim\source\airlib\include\vehicles\configs\px4quadx.hpp:50]
UE4Editor_AirSim!MavMultiRotorConnector::initialize() [c:\users\user\documents\unreal projects\first\plugins\airsim\source\mavmultirotorconnector.cpp:11]
UE4Editor_AirSim!ASimModeWorldMultiRotor::createVehicle() [c:\users\user\documents\unreal projects\first\plugins\airsim\source\simmodeworldmultirotor.cpp:121]
UE4Editor_AirSim!ASimModeWorldMultiRotor::createVehicles() [c:\users\user\documents\unreal projects\first\plugins\airsim\source\simmodeworldmultirotor.cpp:105]
UE4Editor_AirSim!ASimModeWorldMultiRotor::BeginPlay() [c:\users\user\documents\unreal projects\first\plugins\airsim\source\simmodeworldmultirotor.cpp:11]
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_UnrealEd
UE4Editor_UnrealEd
UE4Editor_UnrealEd
UE4Editor_UnrealEd
UE4Editor_UnrealEd
UE4Editor

Timebutt, 2017-03-06:
Hi @g-ch, how are you running your project exactly? Are you starting it from the Visual Studio solution? Have you right-clicked your *.uproject file and selected 'Generate Visual Studio project files' before building the Visual Studio project?

A big help to getting the environment to run correctly are the two videos on the following page:
[https://github.com/Microsoft/AirSim/blob/master/docs/manual_flight.md](https://github.com/Microsoft/AirSim/blob/master/docs/manual_flight.md)

jjy201314, 2017-03-06:
error: Unable to instantiate instance of 'AirSim' object type from compiled assembly 'myprojectModuleRules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.  Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements.  The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: 调用的目标发生了异常。 ---> System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable.  See readme: https://github.com/Microsoft/AirSim/blob/master/docs/install_boost.md	

MSB3075	命令“"D:\wurenji install\UE_4.14\Engine\Build\BatchFiles\Build.bat" myprojectEditor Win64 DebugGame "D:\Unreal Projects\myproject\myproject.uproject" -waitmutex”已退出，代码为 5。请验证您是否有足够的权限来运行此命令。	
 
I have  two errors when debug my myproject, what should  i do,thanks

Timebutt, 2017-03-06:
As the debug printout says, there is something wrong with your Eigen installation. 

Have you installed the Boost and Eigen dependencies, and added them to your environment variables, as instructed [here](https://github.com/Microsoft/AirSim/blob/master/docs/install_boost.md) (Boost) and [here](https://github.com/Microsoft/AirSim/blob/master/docs/install_eigen.md) (Eigen)?

g-ch, 2017-03-07:
@Timebutt thanks a lot. i did not run the project by visual studio. the video helps a lot  :)

Timebutt, 2017-03-07:
Did this fix your problem? If so, you can close down this issue ;)

vpervalla, 2017-03-07:
@g-ch  Can you please tell me precisely how did you solve? Running the project from Visual Studio was enough?
I have the same problem trying to setup the envirnment. I followed the video [Unreal AirSim Setup Video](https://www.youtube.com/watch?v=1oY8Qu5maQQ&feature=youtu.be). Now I am trying right-clicking my *.uproject file and selecting 'Generate Visual Studio project files'  but opening .sln file with Visual Studio and pressing F5 will result in building the project and opening Unreal Engine that crash when I press the play button.

Thanks.

g-ch, 2017-03-07:
nothing is wrong with your procedures. i got the error because i ran the project directly by unreal engine. the problem solved by running the project in visual studio(F5). then the engine started. there is a internal problem when the play button is pressed, but i just ignore it and it goes well. 
did you change the debug mode in visual studio and the world settings in unreal engine? as the video shows

vpervalla, 2017-03-07:
@g-ch you mean change the configuration of my project to debug_game?
Anyway I solved by putting and compiling AirSim in DIR=Users/<user>/Documents. Found that because method createTextFile() mentioned at the beginning (crash screen) tries to open settings.json file in DIR/AirSim
Very likely that I missed some point in the video ;)

lovettchris, 2017-03-07:
Interesting "Users//Documents" would mean that your system has no USERPROFILE environment variable?  I didn't know that was possible on Windows.  Can you open a command prompt and type "set USERPROFILE" and tell me what you see ?  Can you also open "File Explorer", right click on "Documents", select Location and let me know what the location of your Documents folder is set to ?

vpervalla, 2017-03-09:
suppose my username is Foo; typing set USERPROFILE results in this output:
USERPROFILE=C:\Users\Foo
then I opened FileExplorer, right click on Documents (there is no Location thing), Properties and Location/Path, results in:
C:\Users\Foo\Documents

closed 2017-03-11 by g-ch