microsoft/AirSim #85: DroneShell issues

Timebutt, 2017-03-06:
I finally got the entire AirSim environment up and running using PX4 SITL simulation and a PS3 controller (emulated as an Xbox 360 controller). Now I want to programmatically get images and telemetry date from the drone for post processing and development. 

The [instruction video](https://youtu.be/HNWdYrtw3f0?t=513) shows how DroneShell can be used to easily enter and test commands from the AirLib library. This, however, does not work when I try it. 

Here is a screen capture, taken from my DroneShell prompt after entering some commands. 

[image]

As in the video, I start with the _Arm_ command (after a new Unreal simulation start) which does nothing and blocks the command line for 15 seconds (when I arm using my controller, the drone immediately arms). Next, I try the _TakeOff_ command, which again blocks up the command line for 15 seconds (the _wait_max_seconds_ variable) and then spits out '_RPC Error: rpclib: function 'takeoff' (called with 1 arg(s)) threw an exception. The exception contained this information: TakeOff command - timeout waiting for response_.'

One might think there is no connection to the drone, but that is not true. Upon using the _GetImage_ command, I instantly get a response, saying the image is being saved somewhere in the AirSim folder (I verified, the image is also correct). So certainly, DroneShell can correctly connect to the Unreal simulation and retrieve live data from it.

Is something broken in the DroneShell application since the demo video? Am I missing something here?

Here is the content of my settings.json file. Default configuration, except I changed the _UseSerial_ variable to false (as instructed in the [sitld.md](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md) documentation file) 
```
{
  "Pixhawk": {
    "LocalHostIp": "127.0.0.1",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14560,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": false,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  }
}
```

lovettchris, 2017-03-10:
I did some debugging on this and found out that the PX4 SITL works differently than a real Pixhawk hardware.  On real hardware the HIL mode sensor data and the commands to arm and fly the drone can be sent over the same single mavlink connection.  In PX4 SITL mode the Simulator channel can only be used for simulator HIL_SENSOR messages, and cannot be used to send commands, which is what we are trying to do when you use DroneShell.  Some have already noted that QGroundControl however, can control the drone while it is in the AirSim simulator, (same goes for jMAVSim) and this is because QGroundControl is using a different UDP port (14556).  So PX4 SITL mode is listening to different mavlink channels over different UDP ports.  You can see this when you type "mavlink status" from the PX4 SITL terminal.

So I fixed our implementation to allow two different mavlink channels.  To get SITL mode working you have to set these properties in your settings.json:

    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,

see [this checkin](https://github.com/Microsoft/AirSim/commit/a8aabdcba4580796d46440cb5e4f925ce6f11416).

Timebutt, 2017-03-10:
Hi @lovettchris, I managed to control the drone using DroneShell using these adaptations, but only after fiddling around a bit. I found two different scenarios. Let me explain: 

1. I start the simulation, and launch DroneShell. I issue the arm command, nothing happens. I issue _takeoff_, again nothing happens. Then, after _max_timeout_seconds_ I get an RPC error _drone hasn't came [sic] to expected z  of -3.000000 within time 15.000000 sec within error margin 0.500000 (current z = 0.013608)_. If I then issue _arm_ again, the drone will arm and immediately take off.

[image]

2. Another way to make immediate contact with the drone, is when I first request _GetImage_. If I then wait for DroneShell to become ready for the next command, I can issue _arm_ and immediately the drone responds. This also only works if the image has actually been saved: sometimes I get a response saying the image couldn't be saved, and I need to look for errors in the simulator. 

So my feeling is: the connection over port 14556 isn't being set up correctly immediately after launching DroneShell. Somehow, extra input/triggering is required for the actual connection to be set up.

lovettchris, 2017-04-04:
I added some [documentation in sitl.md](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md) under **Check the Home Position** before trying to takeoff, can you check if that is the issue you are running into here?

closed 2017-04-19 by sytelus