microsoft/AirSim #87: Will a ps4 controller work?

cwhil001, 2017-03-07:
I was trying with a ps4 controller but qgroundcontrol was not able to recognize it, is there a way to make it work with a ps4 controller?

Timebutt, 2017-03-07:
Hi @cwhil001, I got AirSim working using a PS3 controller and a PX4 SITL configuration. What I had to do to get my PS3 controller working, is emulate it as a Xbox 360 controller, after which QGroundControl automatically found it. The [x360ce](http://www.x360ce.com/) controller emulator is what I used. 

I had a quick look around, and found that there is an even nicer solution for the PS4 controller, called [DS4Windows](http://ds4windows.com/). Using this solution, it should definitely be possible to setup your configuration.

Please let us know if you get it working, so we can update the documentation. I already submitted a pull request to edit the documentation, confirming that a PS3 controller works. Once you get yours up and running, I'll add the PS4 controller and driver solution.

cwhil001, 2017-03-08:
Im having an issue with the PX4 SITL configuration, when i type make posix_sitl_default in bash i get the following error 
https://cloud.githubusercontent.com/assets/25446938/23722015/b979fa1a-0411-11e7-87f6-8568527b5331.PNG
Copy and paste that link on your address bar to see the screen shot

I was able to download ds4windows and it recognises my ps4 controller but it doesnt work in airsim

cwhil001, 2017-03-08:
I have also installed the xbox360e and it recognises the ps4 controller, but qground control does not find it it doesnt show anything.

lovettchris, 2017-03-08:
Your screen shot shows you are missing 'cmake' tool, please follow the directions here:
[https://dev.px4.io/starting-installing-linux.html](https://dev.px4.io/starting-installing-linux.html).

cwhil001, 2017-03-09:
Hmmm i dont know its working for me, i am actuallydoing this on a windows machine but the instructions said to download bash and then do the part that said ....make posix_sitl_default.
I see that https://dev.px4.io/starting-installing-linux.html has a windows guide so i am installing it that way but i dont understand, does q ground control have to recognize my ps4 controller? So once i do the px4 installation, q ground control will be able to recognize it?

cwhil001, 2017-03-09:
@Timebutt hiw exactly did you configure and build px4 sitl? Because i have tried everything amd nothing seems to be working

Timebutt, 2017-03-09:
Hi @cwhil001, I was able to build PX4 following the instructions provided in the [AirSim documentation](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md#software-in-loop-simulation-sitl), and the [PX4 Devguide](https://dev.px4.io/starting-installing-linux.html#installation). In the PX4 documentation, you must follow all steps in the 'Installation' paragraph. Having completed those, you should be able to build the PX4 firmware. If not, please provide a clear screenshot of the output so we can help you diagnose the issue.

On a side note: your joystick controller (be it PS3, PS4, Xbox 360 ...) will only be recognized once the entire environment is running. So you need to have the PX4 running in bash and the AirSim simulation in Unreal, only then will QGroundControl connect to the virtual vehicle and show you the joystick options.

cwhil001, 2017-03-09:
Okay, I have tried both ways to install, the linux instructions and the windows instructions as I am on a windows machine.

Windows:
For windoes installation on th px4 devguide, I am able to install all the software successfully that shows on this screen shot: 
https://cloud.githubusercontent.com/assets/25446938/23753965/5951ecb8-04a9-11e7-901a-869cd0629eba.png
Then it says continue to run the first build so i click on that link.

I then says to open the px4 console:
https://cloud.githubusercontent.com/assets/25446938/23754120/de13eb40-04a9-11e7-88c3-37d39053e1f2.png

Then:
https://cloud.githubusercontent.com/assets/25446938/23754300/7d265be6-04aa-11e7-8b1c-50ed200303dd.png

Then it says a simulation run is recommended so i click on the link and then when i run make posix_sitl_default jmavsim i get this error:
https://cloud.githubusercontent.com/assets/25446938/23754595/8ba925b2-04ab-11e7-9b76-57b1ad66755f.png

Linux:
I am able to install bash successfully. Then it says "Follow these steps for Linux and follow all the instructions under NuttX based hardware to install prerequisites"

sudo usermod -a -G dialout $USER:
https://cloud.githubusercontent.com/assets/25446938/23753619/26fe2b1a-04a8-11e7-969d-5398f28db818.png

sudo add-apt-repository ppa:george-edison55/cmake-3.x -y:
https://cloud.githubusercontent.com/assets/25446938/23753721/86c9551a-04a8-11e7-9623-ef29499eaad1.png

sudo apt-get update:
https://cloud.githubusercontent.com/assets/25446938/23753784/b8551696-04a8-11e7-894d-72653ab86c32.png

sudo apt-get install python-argparse git-core wget zip \
    python-empy qtcreator cmake build-essential genromfs -y:
https://cloud.githubusercontent.com/assets/25446938/23753896/1cc21a2a-04a9-11e7-906a-95b401fe41a7.png
I get the error that shows on the screen shot: unable to locate package python-empy

Timebutt, 2017-03-09:
It'll be much easier running your PX4 simulator in BashOnWindows. Looks like you're very close, somehow the python-empy package can't be located. 

Since we're using a huge install command, we can maybe track down the issue by using a smaller command. Could you quite simply try the following lines of code, and see if the error remains the same? I verified this on my BashOnWindows, and I don't have an issue locating the package.

```
sudo apt-get update
sudo apt-get install python-empy
```

If the error remains, somehow your bash doesn't update the repositories correctly.

cwhil001, 2017-03-09:
Okay that worked:
https://cloud.githubusercontent.com/assets/25446938/23757269/4928c216-04b4-11e7-9f00-458b554435dd.png

Now do i do the next command the exact way it is or should i shorten it to what? sudo apt-get install ant protobuf-compiler libeigen3-dev libopencv-dev openjdk-8-jdk openjdk-8-jre clang-3.5 lldb-3.5 -y

And after that should i continue to just the section that says NuttX based hardware or do all of them?

Timebutt, 2017-03-09:
Next commands are these:

```
sudo apt-get install python-argparse git-core wget zip python-empy qtcreator cmake build-essential genromfs -y
sudo apt-get install ant protobuf-compiler libeigen3-dev libopencv-dev openjdk-8-jdk openjdk-8-jre clang-3.5 lldb-3.5 -y

sudo apt-get remove modemmanager
sudo apt-get install python-serial openocd flex bison libncurses5-dev autoconf texinfo build-essential libftdi-dev libtool zlib1g-dev  python-empy  -y
sudo apt-get remove gcc-arm-none-eabi gdb-arm-none-eabi binutils-arm-none-eabi gcc-arm-embedded
sudo add-apt-repository --remove ppa:team-gcc-arm-embedded/ppa
```

The last time, your command failed because you typed the '\' character, I think they just used it in their documentation to spread the command over two lines, instead of having one very long one. I remember having an error or warning in one of the last commands, but disregarded it and it all turned out fine ;)

After that you should be able to build PX4.
Let us know how it goes ;)

cwhil001, 2017-03-09:
I get the following error when running that second command you said: 
https://cloud.githubusercontent.com/assets/25446938/23758976/5d78b384-04b9-11e7-9271-96f79d6e800f.png

Unable to locate openjdk-8-jdk and openjdk-8-jre

cwhil001, 2017-03-09:
Nevermind i got it, i just had to do:
sudo add-apt-repository ppa:openjdk-r/ppa
sudo apt-get update
sudo apt-get install openjdk-8-jdk

Its installing i will let you know what happens next

cwhil001, 2017-03-09:
Okay so i was able to run those commands without issues. Now to run the build do i scroll all they wat to the bottom and click the link where it says "Now continue to run the first build"?
https://cloud.githubusercontent.com/assets/25446938/23760290/390907a2-04bd-11e7-8a27-b9a52a5a2b19.png

cwhil001, 2017-03-09:
And do i do the part under "NuttX / Pixhawk based boards"?

cwhil001, 2017-03-09:
Okay so far i was able to actually build px4 and now i get a screen with the giant px4 sign:
https://cloud.githubusercontent.com/assets/25446938/23771975/3a14bdce-04e7-11e7-92ff-6948d4fabb33.png

Im not sure what to do next, i have folowed the instructions but it is not working for me, please help =/

Timebutt, 2017-03-10:
Well, you're basically there. Your PX4 SITL is running, now you just need to connect to it using the Unreal Editor (start simulation, set Game Mode to _SimGameMode_) and check if QGroundControl has established a correct connection to the vehicle. If this is the case, you will be able to configure and use your PS4 controller.

Make sure you have correctly edited your _settings.json_ file (can be found in your AirSim folder) according to [these instructions](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md#software-in-loop-simulation-sitl).

cwhil001, 2017-03-11:
Mmm no i does not seem to be working =/ here are screen shots of everything:

https://cloud.githubusercontent.com/assets/25446938/23819926/f15c2676-05db-11e7-895b-97f7ec0e4062.png

https://cloud.githubusercontent.com/assets/25446938/23819920/d80a894c-05db-11e7-984e-bd98ca3200eb.png

https://cloud.githubusercontent.com/assets/25446938/23819896/7f1e1010-05db-11e7-8ad3-001e32fb7acc.png

https://cloud.githubusercontent.com/assets/25446938/23819902/98c85f66-05db-11e7-8170-197e984f1065.png

https://cloud.githubusercontent.com/assets/25446938/23819907/b3fab4e6-05db-11e7-82c8-b576c3871600.png

https://cloud.githubusercontent.com/assets/25446938/23819912/c43c0756-05db-11e7-87b8-a49ab9132dbc.png

lovettchris, 2017-03-11:
Your settings.json looks different from mine, can you try this instead?

{
  "Pixhawk": {
    "LocalHostIp": "127.0.0.1",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14560,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": false,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  }
}

cwhil001, 2017-03-11:
Okay i am pressing buttons on the ps4 controller and moving joysticks but the drone is not moving. I dont see any errors in the simulator and the terminal looks like it is running things
https://cloud.githubusercontent.com/assets/25446938/23820197/8203c54e-05e1-11e7-9aa0-d7ac2ee391e5.png

And qgroundcontrol does recognize the vehicle:
https://cloud.githubusercontent.com/assets/25446938/23820208/d41d1a7e-05e1-11e7-904e-6c8c24a5df73.png

sytelus, 2017-03-16:
Using gamepad with SITL had been tricky. Have you seen the instructions at the end of this doc? https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md

cwhil001, 2017-03-18:
Yeah i did them but its not working for me =\

closed 2017-04-19 by sytelus

LuoShiLiang, 2019-10-25:
> 我尝试使用ps4控制器，但qgroundcontrol无法识别它，是否有办法使其与ps4控制器一起使用？

> 控制器，但qgroundcontrol无法识别它，是否有办法

Hello ~try this way to use PS4 controller. 
On Windows10,download the "DS4",and use the default  config.
[image]

 Now your PC should have a simulated Xbox 360 controller ,and you can find it on Device manager
[image]

Control pannel->device and printer,find your "Xbox 360 Controller for Windows" ,right click it ->Game controller settings->Advance->Preferred device->"Select XBOX 360 For Windows"->Go UE4 Project to test ~
[image]

LuoShiLiang, 2019-10-25:
> 我尝试使用ps4控制器，但qgroundcontrol无法识别它，是否有办法使其与ps4控制器一起使用？

I use the machine translation tool to translate Chinese into English. If there is confusion in reading, please contact me, I will answer in more detail.