microsoft/AirSim #88: Is there an API document in airsim like Java API  document？

BlackFH, 2017-03-07:


lovettchris, 2017-03-07:
Not complete documentation, but there is a [quick into here](https://github.com/Microsoft/AirSim/blob/master/docs/apis.md) with links to the most important C++ header, namely [DroneControllerBase.hpp](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/controllers/DroneControllerBase.hpp).

BlackFH, 2017-03-07:
After analyzing build.cmd, compiling AirSim is actually copying some imported headers and source files into Unreal\Plugins. Could we just code in UNREAL project which imported from AirSim? If I need to add some control logic in my own project, I really can't find which line to start. (oh, hard to code without API documentation. Especially for a college student like me. XD.)

lovettchris, 2017-03-07:
Yes, I agree it is a bit hard to change AirSim from inside Unreal project because of the way we copy the binaries over from AirSim.    But the source you can modify in Unreal project is the source you see in Plugins\AirSim\Source, that is NOT in the AirLib folder.  This source contains the entry point for our Plugin in the Unreal world.  For example, MavMultiRotorConnector::beginPlay() is what is called when you hit the Play button in the Unreal editor.  

Now the other way you can play is to modify DroneShell, which can connect to the Unreal app while it is running and talk to it over an RPC channel.  If you want to experiment with the DroneControlBase API, that would be an easier way to do it that trying to do it inside the Unreal app.  I added lots of comments to DroneControlBase.hpp :-)

closed 2017-03-08 by BlackFH