microsoft/AirSim #90: Building on linux

andreanicastro, 2017-03-07:
Hi, I am trying to use it on Ubuntu 16.04
I have compiled and installed Unreal Engine using this [guide](https://wiki.unrealengine.com/Building_On_Linux).
Then I followed the [instruction](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md) to build your software on linux.
Finally I run the following commands:
`mkdir -p AirLib/deps/MavLinkCom AirLib/deps/rpclib/lib AirLib/lib
cp -p cmake/output/lib/libAirSim-rpclib.a AirLib/deps/rpclib/lib
cp -p cmake/output/lib/libAirLib.a AirLib/lib
cp -rp MavLinkCom/include AirLib/deps/MavLinkCom
cp -rp cmake/output/lib/libMavLinkCom.a AirLib/deps/MavLinkCom/lib
cp -rp AirLib Unreal/Plugins/AirSim/Source
`
and copy the plugin folder in the project.
However whenever I try to open the project with the editor I get the following message

> Project files could not be generated. Log output:

Running /home/anicastro/software/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe  -makefile -kdevelopfile -qmakefile -cmakefile -codelitefile  -project="/home/anicastro/Documents/Unreal Projects/AirSimTest 4.14/AirSimTest.uproject" -game -engine -progress
Discovering modules, targets and source code for project...
ERROR: Unable to instantiate instance of 'AirSim' object type from compiled assembly 'AirSimTestModuleRules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.  Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements.  The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable.  See readme: https://github.com/Microsoft/AirSim/blob/master/docs/install_boost.md
  at AirSim.AddEigenDependency () <0x4068f400 + 0x0011f> in <filename unknown>:0 
  at AirSim..ctor (UnrealBuildTool.TargetInfo Target) <0x4068f100 + 0x0021b> in <filename unknown>:0 
  at (wrapper managed-to-native) System.Reflection.MonoCMethod:InternalInvoke (System.Reflection.MonoCMethod,object,object[],System.Exception&)
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) <0x7f7d5e9318d0 + 0x0003f> in <filename unknown>:0 
  --- End of inner exception stack trace ---
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) <0x7f7d5e9318d0 + 0x00079> in <filename unknown>:0 
  at System.Reflection.MonoCMethod.DoInvoke (System.Object obj, BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) <0x7f7d5e931760 + 0x000d6> in <filename unknown>:0 
  at System.Reflection.MonoCMethod.Invoke (BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) <0x7f7d5e931980 + 0x00036> in <filename unknown>:0 
  at System.RuntimeType.CreateInstanceImpl (BindingFlags bindingAttr, System.Reflection.Binder binder, System.Object[] args, System.Globalization.CultureInfo culture, System.Object[] activationAttributes, System.Threading.StackCrawlMark& stackMark) <0x7f7d5e77ec10 + 0x005c7> in <filename unknown>:0 
Discovering modules, targets and source code for project...
ERROR: Unable to instantiate instance of 'AirSim' object type from compiled assembly 'AirSimTestModuleRules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.  Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements.  The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable.  See readme: https://github.com/Microsoft/AirSim/blob/master/docs/install_boost.md
  at AirSim.AddEigenDependency () <0x4068f400 + 0x0011f> in <filename unknown>:0 
  at AirSim..ctor (UnrealBuildTool.TargetInfo Target) <0x4068f100 + 0x0021b> in <filename unknown>:0 
  at (wrapper managed-to-native) System.Reflection.MonoCMethod:InternalInvoke (System.Reflection.MonoCMethod,object,object[],System.Exception&)
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) <0x7f7d5e9318d0 + 0x0003f> in <filename unknown>:0 
  --- End of inner exception stack trace ---
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) <0x7f7d5e9318d0 + 0x00079> in <filename unknown>:0 
  at System.Reflection.MonoCMethod.DoInvoke (System.Object obj, BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) <0x7f7d5e931760 + 0x000d6> in <filename unknown>:0 
  at System.Reflection.MonoCMethod.Invoke (BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) <0x7f7d5e931980 + 0x00036> in <filename unknown>:0 
  at System.RuntimeType.CreateInstanceImpl (BindingFlags bindingAttr, System.Reflection.Binder binder, System.Object[] args, System.Globalization.CultureInfo culture, System.Object[] activationAttributes, System.Threading.StackCrawlMark& stackMark) <0x7f7d5e77ec10 + 0x005c7> in <filename unknown>:0 
Discovering modules, targets and source code for project...
ERROR: Unable to instantiate instance of 'AirSim' object type from compiled assembly 'AirSimTestModuleRules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.  Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements.  The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable.  See readme: https://github.com/Microsoft/AirSim/blob/master/docs/install_boost.md
  at AirSim.AddEigenDependency () <0x4068f400 + 0x0011f> in <filename unknown>:0 
  at AirSim..ctor (UnrealBuildTool.TargetInfo Target) <0x4068f100 + 0x0021b> in <filename unknown>:0 
  at (wrapper managed-to-native) System.Reflection.MonoCMethod:InternalInvoke (System.Reflection.MonoCMethod,object,object[],System.Exception&)
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) <0x7f7d5e9318d0 + 0x0003f> in <filename unknown>:0 
  --- End of inner exception stack trace ---
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) <0x7f7d5e9318d0 + 0x00079> in <filename unknown>:0 
  at System.Reflection.MonoCMethod.DoInvoke (System.Object obj, BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) <0x7f7d5e931760 + 0x000d6> in <filename unknown>:0 
  at System.Reflection.MonoCMethod.Invoke (BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) <0x7f7d5e931980 + 0x00036> in <filename unknown>:0 
  at System.RuntimeType.CreateInstanceImpl (BindingFlags bindingAttr, System.Reflection.Binder binder, System.Object[] args, System.Globalization.CultureInfo culture, System.Object[] activationAttributes, System.Threading.StackCrawlMark& stackMark) <0x7f7d5e77ec10 + 0x005c7> in <filename unknown>:0 
Discovering modules, targets and source code for project...
ERROR: Unable to instantiate instance of 'AirSim' object type from compiled assembly 'AirSimTestModuleRules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.  Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements.  The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable.  See readme: https://github.com/Microsoft/AirSim/blob/master/docs/install_boost.md
  at AirSim.AddEigenDependency () <0x4068f400 + 0x0011f> in <filename unknown>:0 
[... cut ...]
  at System.Reflection.MonoCMethod.Invoke (BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) <0x7f7d5e931980 + 0x00036> in <filename unknown>:0 
  at System.RuntimeType.CreateInstanceImpl (BindingFlags bindingAttr, System.Reflection.Binder binder, System.Object[] args, System.Globalization.CultureInfo culture, System.Object[] activationAttributes, System.Threading.StackCrawlMark& stackMark) <0x7f7d5e77ec10 + 0x005c7> in <filename unknown>:0

saihv, 2017-03-07:
There seems to be an issue in how the Unreal building process recognizes the Eigen and Boost paths: even though the environment variables are set, it doesn't find the path. Please refer to the first comment in this thread for a fix. 

https://github.com/Microsoft/AirSim/issues/69

andreanicastro, 2017-03-07:
Thanks for the answer. Unfortunately that did not solve the issue. I still get the message

> 
>     The following modules are missing or built with a different engine version:
> 
>     libUE4Editor-AirSimTest.so
>     libUE4Editor-AirSim.so
> 
>     Would you like to rebuild them now?
> 
But if I agree to do so, the build fails.

saihv, 2017-03-07:
You'll be able to find the reason why it fails in the log in the terminal. Could you check exactly what error it is you're getting?

andreanicastro, 2017-03-07:
This is the only thing that is shown regarding the project before I am asked to rebuild it is 
> LogPluginManager: Loaded Plugin AirSim, From /home/anicastro/Documents/Unreal Projects/AirSimTest/Plugins/AirSim/AirSim.uplugin
> LogInit:Warning: Incompatible or missing module: libUE4Editor-AirSimTest.so
> LogInit:Warning: Incompatible or missing module: libUE4Editor-AirSim.so

After answering yes to the prompt I see

> Using clang version '3.8.0' (string), 3 (major), 8 (minor), 0 (patch)
> ERROR: Unable to instantiate instance of 'AirSim' object type from compiled assembly 'AirSimTestModuleRules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.  Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements.  The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable.  See readme: https://github.com/Microsoft/AirSim/blob/master/docs/install_boost.md
>   at AirSim.AddEigenDependency () <0x40a53460 + 0x00117> in <filename unknown>:0 
>   at AirSim..ctor (UnrealBuildTool.TargetInfo Target) <0x40a53160 + 0x0021b> in <filename unknown>:0 
>   at (wrapper managed-to-native) System.Reflection.MonoCMethod:InternalInvoke (System.Reflection.MonoCMethod,object,object[],System.Exception&)
>   at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) <0x7f6a135318d0 + 0x0003f> in <filename unknown>:0 
>   --- End of inner exception stack trace ---
>   at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) <0x7f6a135318d0 + 0x00079> in <filename unknown>:0 
>   at System.Reflection.MonoCMethod.DoInvoke (System.Object obj, BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) <0x7f6a13531760 + 0x000d6> in <filename unknown>:0 
>   at System.Reflection.MonoCMethod.Invoke (BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) <0x7f6a13531980 + 0x00036> in <filename unknown>:0 
>   at System.RuntimeType.CreateInstanceImpl (BindingFlags bindingAttr, System.Reflection.Binder binder, System.Object[] args, System.Globalization.CultureInfo culture, System.Object[] activationAttributes, System.Threading.StackCrawlMark& stackMark) <0x7f6a1337ec10 + 0x005c7> in <filename unknown>:0 
> LogInit:Warning: Still incompatible or missing module: libUE4Editor-AirSimTest.so
> LogInit:Warning: Still incompatible or missing module: libUE4Editor-AirSim.so
> 

>

saihv, 2017-03-07:
You're still getting the same error, which means the eigen_path string was not set correctly.

>  System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable. See readme: 

Did you download Eigen and Boost into new folders as described in these documents? 

https://github.com/Microsoft/AirSim/blob/master/docs/install_eigen.md
https://github.com/Microsoft/AirSim/blob/master/docs/install_boost.md

lovettchris, 2017-03-25:
closing now that there are new [Linux Build Instructions](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md) and Linux build is passing in Travis CI.

closed 2017-03-25 by lovettchris