microsoft/AirSim #91: Switching to first person view affects the scene/depth/segmentation camera

saihv, 2017-03-07:
When I am flying the drone in the fly with me mode, the three views (depth, scene, segmentation) on the bottom usually stream images from the drone's PoV as expected. But when I switch to first person view while flying, it's as if those views are now streaming something different, like a side view: in fact, I can even see the drone itself in those views. 

[image]

lovettchris, 2017-03-07:
Yes, they switch with the main view, so when the main view is FPV they become the chase camera, when you switch main view to chase camera, they become FPV.  We probably should provide separate control over the camera used in those views...

saihv, 2017-03-07:
Ah, sorry, didn't know this was expected. It would be nice to be able to pick between the views for flexibility. Closing this for now.

closed 2017-03-07 by saihv