microsoft/AirSim #92: Possibility of simulating more than one vehicle

saihv, 2017-03-08:
I was wondering if it's possible or supported to simulate more than one quadrotor, where each can be controlled by a separate Pixhawk (or even an SITL instance). As a first step, I was able to add two actors to the level blueprint, so two players (drones) are spawned, but before I delve into the code further, I was hoping to get some thoughts on whether this is doable. 

Looks like the `ASimModeWorldMultiRotor::createVehicles` method has support for more than one vehicle, where the method looks for all existing 'flying pawns' and then assigns `fpv_vehicle_connector_` to the active flying pawn, would it be possible to have multiple such `VehicleConnectorBase` connectors?

sytelus, 2017-03-09:
Yes, it is possible to have multiple vehicles in AirSim. The biggest issue you would encounter right now is how to set up PX4. If you are up for the task then you can look up in json that has settings per vehicle and each vehicle is assigned a name. So if you have two instances of FlyingPawn, you can assign different VehicleName in those instances and then put settings for them in json.

The obvious issue in going above route is PX4. So we are hard at work to eliminate dependency on PX4. In our next version, we want to integrate some form of firmware right in AirSim so there is no real hardware or  even SITL required. Once that happens, it should be much easier to do multiple vehicles.

closed 2017-03-09 by sytelus

saihv, 2017-03-19:
@sytelus I was wondering if you'd have any advice about multiple vehicles with regards to the new version. I tried duplicating the BP_FlyingPawn asset in order to create two players, and also duplicated the blueprint for spawning. I assigned the RosFlight type to them in the code so that they wouldn't need Pixhawks connected etc. But the blueprint complains that only one "player controller" is present

`Accessed None trying to read property Callfunc_GetPlayerController_ReturnValue`

and so one of the quadrotors is created and placed correctly, and I can see the model for the other one in the world, but it usually does something weird like falling through the ground (i.e., not recognizing any collisions etc.).