microsoft/AirSim #95: Can't grab more than one type of image through API.

micmancini16, 2017-03-09:
Hi,

I'm trying to write a code which grabs simultaneously depth, segmentation and scene images on a running Unreal simulation. 
My `main()` method implements the following pseudo-code:

```
while (simulation_active) 
{ 
   //these methods would run on separate threads
   grabDepth();
   grabSegmentation();
   grabScene();

   //do something with the images

   sleep_for_a_while(); //e.g. 50 ms
}
```
The `grab<ImageType>` methods use AirSim API functions to contact RPC server running on Unreal and grab images from the simulated cameras

```
grab<ImageType>()
{
        //client is a msr::airlib::RpcLibClient object
        client.setImageTypeForCamera(camera_id, DroneControllerBase::ImageType::<ImageType>);
        std::vector<uint8_t> image = client.getImageForCamera(camera_id,DroneControllerBase::ImageType::<ImageType>);
        
         if (image.size() > 1)
                //Image is successfully grabbed. Do something
}
```
I have success when calling only one of the above-mentioned single `grab<ImageType>` inside my 'while' cycle. Instead, my code fails when trying to call simultaneously different `grab<ImageType>` methods. Only one type of image is successfully grabbed, while other methods fail the `image.size() > 1` check. I run into this issue either when I call `grab` methods on separate threads or when I call them in a single-threaded fashion. 

I suppose that one of the issues might be that the only `camera_id` value which seems to be valid is 0 and the server is not able to manage different `<ImageType>` requests on the same camera. However, I don't know how to add different working cameras on the simulation, so that I could try to use, for example, camera 0 for depth images, camera 1 for segmentation and so on.
I tried creating different RpcLibClient object for each `grab<ImageType>` methods, but that does not seem to solve the problem.

Any help is appreciated!

lovettchris, 2017-03-09:
You correctly deduced that you need to use a different camera id fro each type of picture you want to get back.  AirSim currently supports up to 3 cameras with id=0,1,2, and these map to three little views you see across the bottom of the screen.  Note that if you hide those views then those cameras will not work.

micmancini16, 2017-03-10:
I already tried to set up different camera id's for different ImageTypes but I had no success on doing it. Only camera_id = 0 seems to work, regardless of the selected ImageType. I start Unreal simulation using default camera configuration: 3rd person camera view on the quadrotor and all the three little views on the bottom of the screen.

lovettchris, 2017-03-10:
Ah, sometimes there is a delay between when you request a given image type via setImageTypeForCamera, and when he image is actually available via getImageForCamera so try calling getImageForCamera  in a retry loop...

micmancini16, 2017-03-10:
I moved the `setImageTypeForCamera` calls out of the grab<ImageType> methods, placing them just before the main loop. I checked the assignment success by calling the `getImageTypeForCamera(int camera_id)` for each id and everything looks fine.

Digging into the implementation of the `DroneControllerBase::getImageForCamera(int camera_id, ImageType type)` method, I realized that it fails to find camera_id 1 and 2 and so it returns an empty vector. (Code-wise, it fails the `if (it != images.end())` condition; I pasted the API implementation here as reference)

```
vector<uint8_t> DroneControllerBase::getImageForCamera(int camera_id, ImageType type)
{
    StatusLock lock(this);

    //TODO: bug: MSGPACK bombs out if vector if of 0 size
    static vector<uint8_t> empty_vec(1);

    vector<uint8_t> result;

    //TODO: perf work
    auto it = images.find(camera_id);
    if (it != images.end()) {
        auto it2 = it->second.find(type);
        if (it2 != it->second.end())
            result = it2->second;
        else
            result = empty_vec;

    } else
        result = empty_vec;

    if (result.size() == 0) {
        result = empty_vec;
    }

    return result;
}
```
I'll keep digging into the method implementations to find out more about this bug.

micmancini16, 2017-03-13:
UPDATE: The problem lies on the implementation of the [ASimModeWorldMultiRotor ](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/SimModeWorldMultiRotor.cpp) class. Its`Tick()` method actually handles solely camera 0 when it exports Unreal-generated textures to the [DroneControllerBase](https://github.com/Microsoft/AirSim/blob/master/AirLib/src/controllers/DroneControllerBase.cpp) class (lines 30-50), causing all the above-mentioned problems. 
Parameterizing those code lines with respect to different camera_id's solved the problem.  

Nevertheless, I still need to work on a more efficient solution, as, for now, my code is quite rough and I experience low framerates.

lovettchris, 2017-03-13:
Oh, you are correct, I will get that fixed, we have some ideas on how to improve performance, for example, the image PNG compression could move to a background thread...  How many images per second are you trying to get ?

micmancini16, 2017-03-14:
I'm trying to capture images with a rate of 10 FPS. Simulation lags at about 1-2 FPS when running my code. I guess it's the same problem other people experienced when using the log recording functionality (which I experienced too).

lovettchris, 2017-03-14:
Ok, we'll make this a work item to find a way to increase image capture FPS.

Dynode, 2017-05-16:
@micmancini16 Would you share the fix you mentioned on Mar 13th? I'm trying to grab screenshots from the Scene PIP while looking in FPV.

micmancini16, 2017-05-16:
@Dynode I edited the Tick() method of the [https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/SimMode/SimModeWorldMultiRotor.cpp](SimModeWorldMultiRotor) class, replicating the code in lines 114-134 for camera 1 and 2. Then, you should be able to get all PIPs simultaneously using the API as explained in the [https://github.com/Microsoft/AirSim/blob/master/docs/apis.md](documentation).

However, I need to clarify that I'm not currently using the most recent version of the repository as I found a sweet spot for my work development with an older version and stuck with that. I don't know if this would work with the newer versions.

Dynode, 2017-05-17:
@micmancini16 Aha, I see. Thanks, I'll play around with the code you refer to.

lovettchris, 2017-06-10:
Ok, this should be fixed now.  I simplified the camera api, (getImageTypeForCamera , setImageTypeForCamera are gone).  So you can now fetch all three camera types using python and render them simultaneously, see [camera_views](https://github.com/Microsoft/AirSim/blob/master/docs/camera_views.md#performance) for performance info on that.

closed 2017-06-10 by lovettchris