microsoft/AirSim #96: Unable to arm or control drone once simulation starts

inkjet, 2017-03-09:
Thanks to everyone's help, I'm now up running with Ubuntu 16.04. But I can't seem to fly the drone once I start SimGameMode -- it just tumbles to the ground because I'm unable to arm or control it.  

I've followed the procedure to install PX4 and QFlightControl, and it appears I was able to set up my Logitech joystick as a HIL Quadcopter X as mentioned in [the tutorial video](https://www.youtube.com/watch?v=HNWdYrtw3f0&feature=youtu.be).  But I can't seem to arm the drone for flight.

I've also tried using "Arm" command via the DroneShell binary included with the AirSim build, but that also does nothing. Any idea what I can do finally get up and running?

Screenshot of my environment:

[image]

Timebutt, 2017-03-09:
Hi @inkjet, seems the connection to both PX4 and QGroundControl is established correctly. Quick test to see if you can control the drone using QGroundControl would be to select the 'paper plane' button (fourth button on the button ribon in QGroundControl) and use the option 'arm' at the bottom to arm the drone, see image:

[image]

This way, we can verify that you can effectively send command to Unreal simulation. Don't worry about DroneShell, I myself can't get that to work right now and have opened an issue (see #85).

Next up: have you configured a button on your controller to execute the arm command? In the following screenshot you can see how I have it set up for my PS3 controller, maybe there's just no button mapped to the arm command in your configuration.

[image]

inkjet, 2017-03-09:
@Timebutt 

Thank you for the info.  I am able to arm the "paper plane" in QGroundControl.  I've also mapped Arm and Disarm commands to my joystick.  

But I am still stuck when I return to the AirSim environment.  Whenever I press the "arm" button I see `INFO  [tone_alarm] positive` in the PX4 terminal and `QObject::startTimer: Timers cannot be started from another thread` in  QGroundControl terminal.  (Example below.)  

[image]

It seems like the PX4 sees the joystick, it's just not connecting with AirSim.

lovettchris, 2017-03-10:
The original bug reported here to do with DroneShell is now fixed.  Turns out that the PX4 SITL works differently than a real Pixhawk hardware.  On real hardware the HIL mode sensor data and the commands to arm and fly the drone can be sent over the same single mavlink connection.  In PX4 SITL mode the Simulator channel can only be used for simulator HIL_SENSOR messages, and cannot be used to send commands, which is what we are trying to do when you use DroneShell.  Some have already noted that QGroundControl however, can control the drone while it is in the AirSim simulator, (same goes for jMAVSim) and this is because QGroundControl is using a different UDP port (14556).  So PX4 SITL mode is listening to different mavlink channels over different UDP ports.  You can see this when you type "mavlink status" from the PX4 SITL terminal.

So I fixed our implementation to allow two different mavlink channels.  To get SITL mode working you have to set these properties in your settings.json:

    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,

see [this checkin](https://github.com/Microsoft/AirSim/commit/a8aabdcba4580796d46440cb5e4f925ce6f11416).

inkjet, 2017-03-10:
@lovettchris 

I'm still having the same issues getting the drone to arm.  I downloaded and compiled the latest checkin, and when I execute the "Arm" command from DroneShell, it just freezes.  Here is my settings.json file -- any clue as to what I may be missing?

[settings.json.txt](https://github.com/Microsoft/AirSim/files/835020/settings.json.txt)

lovettchris, 2017-03-10:
Here's mine, note the sitl port is different ( you can also delete the old stuff outside the "Pixhawk", those are no longer used):

```
{
  "Pixhawk": {
    "LocalHostIp": "127.0.0.1",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": false,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  }
}
```

inkjet, 2017-03-11:
@lovettchris Thank you for sharing your settings. I'm still not having luck controlling via DroneShell.  The "arm" command doesn't give an error, but nothing happens in the sim.  This is despite seeing the `Take off to 3.000000` in the Unreal Editor terminal:

Here is the my history in DroneShell:

`
==||=> Pos
==||=> Local position: x=0, y=0, z=0
                                    Global position: lat=0, lon=0, alt=0
==||=> Arm
==||=> TakeOff
RPC Error: rpclib: function 'takeoff' (called with 1 arg(s)) threw an exception. The exception contained this information: TakeOff command - timeout waiting for response.
==||=> GetImage
==||=> error getting image, check sim for error messages
`

inkjet, 2017-03-12:
@lovettchris Update - the inability to arm was with a previous version.  I did a fresh compile with [ce6f11416](https://github.com/Microsoft/AirSim/commit/a8aabdcba4580796d46440cb5e4f925ce6f11416) and now the sim simply crashes.  I am able to load the Unreal Editor, start SimGameMode, and after seeing the drone for a few seconds, the Editor hard quits.  The error repeated several times in the terminal is:

```
MavLinkConnection subscriber threw exception: std::exception
Exception occurred while updating world: UdpClientPort socket send failed with error: 111
```

I am using the same `settings.json` file you provided above.  Attached is the full dump from the terminal.  

[airsim_core_dump.txt](https://github.com/Microsoft/AirSim/files/836261/airsim_core_dump.txt)

(Side note - when I used the previous version, the value for `SitlPort` would get changed to 14560 when the sim would load no matter its initial value, which probably lead to the inability to arm errors mentioned above.)

lovettchris, 2017-03-13:
Thanks for the full error report, that is handy.  The error 111 is ECONNREFUSED, meaning the target refused to connect.  I added some better error handling (it should not have crashed the simulator).  But I think the fix in this case might be to set the following to an empty string:

"LogViewerHostIp": "",

This will stop it from trying to write telemetry to the LogViewer, since you probably have no log viewer listening.  But I'm a bit confused why that would generate a ECONNREFUSED, the UDP message should be a fire and forget if no one is listening, at least that's what it does on windows.  So you might also have a firewall issue or something...

[see fix](https://github.com/Microsoft/AirSim/commit/29b6d2279450fbf192f0bb914c6a4a8c1da63e85)

closed 2017-03-16 by sytelus