microsoft/AirSim #97: Error with buld.cmd  eigen3\eigen\src\core\arch\cuda\half.h : error C2220

whyscience, 2017-03-09:
error eigen-3.2.2\eigen3\eigen\src\core\arch\cuda\half.h   
warning C4819: 
error C2220:
[image]

ignore the Chinese

whyscience, 2017-03-09:
problem sovled
open eigen3\Eigen\src\Core\arch\CUDA\half.h and save using UTF-8 coding

closed 2017-03-09 by whyscience

Focke-Wulf, 2017-03-19:
666666 Thanks for the Solution!

woooha, 2017-06-26:
You'd better to save file with UTF-8 with BOM encoding.

chhtz, 2017-07-19:
FYI: This had been fixed upstream a while ago in Eigen by removing the non-ASCII quotation marks: https://bitbucket.org/eigen/eigen/commits/54f26e677873161213620be9d5f4af22cab94c27

qixuxiang, 2018-01-27:
I sovled the problem in this way:

[image]