microsoft/AirSim #99: Expose lowest-level controls through public API

jpreiss, 2017-03-09:
The [`DroneControlBase`](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/controllers/DroneControllerBase.hpp) public API only allows control via higher-level commands like attitude+z, velocity vector, or following a waypoint path. Meanwhile, the true control inputs to a quadrotor are 4 motor voltages, or at a slightly higher level, angular velocity and thrust, aka acro/rate mode.

By offering only the higher-level interface, AirSim makes any autonomy-level control scheme dependent on the quality of the low-level controller that translates the high-level commands into the true system inputs. This could be a big problem for fast flight, where many of the simplifying assumptions in typical low-level controllers become invalid.

It would be great if the lowest-level inputs would be exposed, so users can take advantage of the physics simulation and graphics engine without depending on a particular controller implementation.

Btw, I am happy to work on this if you are accepting pull requests.

sytelus, 2017-03-10:
There are technical limitations on exposing very low level functionality through APIs because its difficult to make APIs cannot work at more than 50Hz rate and also they are not real-time (i.e. there would be significant variance in timing between call to call). This is why APIs focus is on high level functionality. However, we are planning to implement RC interface for API which means you would be able to supply RC inputs to simulator via API at ~50Hz rate. We already have angle controls via API as you have mentioned. These two should deliver most functionality that people would need. It would be easy to implement rate control via API as well but we will leave that to community :).

If you need lower level control than that (for example, direct motor controls) then you should ideally write a controller (i.e. implement DroneControllerBase) so your code can run at rate as high as 500Hz in high priority thread.

closed 2017-04-19 by sytelus

jpreiss, 2017-10-06:
I am interested in this issue again. Now that `simple_flight` exists, it is my understanding that we have ability to single-step the main loops of both simulator and flight controller, i.e. we are never dependent on the real-life wall clock time. Therefore, we should no longer be sensitive to latency/bandwidth/jitter of the interprocess communication between API client and AirSim. Does this mean it is now viable to add lower level control options to the public API? Again, I am willing to do this work myself if it will be accepted into mainline.

zbenic, 2018-01-15:
I am also very interested in getting the access to the low-level quad controls.
Would very much like to see @jpreiss latest question answered.