microsoft/AirSim #100: refresh visual studio project failed

liuli2008212, 2017-03-10:
When I refresh visual studio project at unreal engine. Happened this issue as below and cann't be updated :

There was an error trying to generate the project file.

Running F:/airsim/UE_4.14/Engine/Binaries/DotNET/UnrealBuildTool.exe  -projectfiles -project="F:/airsim/flyUeproject/airsim/airsim.uproject" -game -rocket -progress
Discovering modules, targets and source code for project...
ERROR: Unable to instantiate instance of 'AirSim' object type from compiled assembly 'airsimModuleRules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.  Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements.  The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: ???õ?Ŀ?귢?????쳣?? ---> System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable.  See readme: https://github.com/Microsoft/AirSim/blob/master/docs/install_eigen.md
   ?? AirSim.AddEigenDependency() λ?? f:\airsim\flyUeproject\airsim\Plugins\AirSim\Source\AirSim.Build.cs:?к? 77
   ?? AirSim..ctor(TargetInfo Target) λ?? f:\airsim\flyUeproject\airsim\Plugins\AirSim\Source\AirSim.Build.cs:?к? 65
   --- ?ڲ??쳣??ջ???ٵĽ?β ---
   ?? System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   ?? System.Reflection.RuntimeConstructorInfo.Invoke(BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   ?? System.RuntimeType.CreateInstanceImpl(BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes, StackCrawlMark& stackMark)
   ?? System.Activator.CreateInstance(Type type, BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes)
   ?? System.Activator.CreateInstance(Type type, Object[] args)
   ?? UnrealBuildTool.RulesAssembly.CreateModuleRules(String ModuleName, TargetInfo Target, FileReference& ModuleFileName)
[image]

Please help me.Thank you very much!

clovett, 2017-03-10:
Did you follow the instructions in the error message?  Namely, https://github.com/Microsoft/AirSim/blob/master/docs/install_eigen.md

liuli2008212, 2017-03-13:
clovett,thank you.I have built succeed as the instructions of install_eigen.md,but refreshing visual studio project has been failed.

lovettchris, 2017-03-13:
Did you remember to set the EIGEN_ROOT environment variable, then restart Visual Studio ?

[image]

closed 2017-03-16 by sytelus